LumaTeam/Luma3DS #1089: Injecting cias with GM9 crashes after shutdown

YYoshi241, 2018-05-30:
Random Info
**System model:**
O3ds XL

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

 Boot9Strap 1.3

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: ( none)

PIN lock: ( none)

Autoboot EmuNAND: ( nope)

Use EmuNAND FIRM if booting with R: ( no)

Enable loading external FIRMs and modules: (no )

Enable game patching: (no )

Show NAND or user string in System Settings: (yes )

Show GBA boot screen in patched AGB_FIRM: ( yes)

Patch ARM9 access: ( forced, Luma 9.0)

Set developer UNITINFO: (no )

Disable ARM11 exception handlers: (yes )

--

**Explanation of the issue:**
Injecting CIAs into H&S w/ GM9 causes the An exception error screen when rebooting/power ing after being ran. (Doesn't matter the GM9 Ver.)
Only occurs on Luma 9.0 not 8.1.1 

**Steps to reproduce:**

1.inject cia into h&s

2.run the h&s app

3. Power off or reboot any way

I do not have a dump at this time but it should be pretty easy to reproduce.

Shadow322, 2018-05-30:
We will not help when a user disables arm11 exception handlers

YYoshi241, 2018-05-31:
This is not related to ARM11, it crashes with an ARM9 exception.

Hikari-chin, 2018-05-31:
Ticking that option invalidates your eligibility to submit issues as a whole, but this is unlikely to be a Luma issue anyway. CIA injection into existing apps was always an extremely dirty/hacky way to do things and is highly not recommended in light of superior options.

closed 2018-05-31 by Hikari-chin