LumaTeam/Luma3DS #1090: Latest devkitPro/devkitARM (?) breaks Luma compilation

NWPlayer123, 2018-06-04:
I decided to uninstall and reinstall devkitPro a few hours ago, I don't think there are any config issues, just upgrading to GCC 8.1 broke everything. It gets through the rest of the compilation, just breaks on k11_extension for whatever reason.
```
C:\Users\NWPLAY~1\AppData\Local\Temp\ccMPsFsmdebugobjtem: file not recognized: File truncated
collect2.exe: error: ld returned 1 exit status
lto-wrapper.exe: fatal error: C:\devkitPro\devkitARM\bin\arm-none-eabi-gcc.exe returned 1 exit status
compilation terminated.
c:/devkitpro/devkitarm/bin/../lib/gcc/arm-none-eabi/8.1.0/../../../../arm-none-eabi/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
make[1]: *** [Makefile:36: build/k11_extension.elf] Error 1
make[1]: Leaving directory '/c/Hacks/Tools/git/Luma3DS/k11_extension'
make: *** [Makefile:125: build/k11_extension.bin] Error 2
```

profi200, 2018-06-04:
That seems to be a gcc 8.1 bug. Someone else had the same problem. The workaround is using devkitARM within WSL or on Linux.

fincs, 2018-06-04:
Or disable lto.

yuukieve, 2018-06-12:
Per @profi200's comment, seems to be a gcc 8.1 bug rather than an issue with Luma itself, and it doesn't seem to effect the Linux dkA.

closed 2018-06-12 by yuukieve

NWPlayer123, 2018-06-12:
mind elaborating what bug, for future reference? @yuukieve

Jerry-Shaw, 2018-06-29:
So, is there anyway to get it built under the latest devkitARM? Seems like I cannot have the old version downloaded and installed.

NWPlayer123, 2018-06-29:
as fincs suggested, just remove "-flto" (link-time optimization) from https://github.com/AuroraWright/Luma3DS/blob/master/k11_extension/Makefile#L17 @Jerry-Shaw

Jerry-Shaw, 2018-06-29:
@NWPlayer123 , @fincs 
I removed "-flto" from the main Makefile and failed. Never thought to remove it from another Makefile.
Will try it later.
Thanks so much.

zobland, 2018-10-07:
as fincs suggested, just remove "-flto" (link-time optimization) from 

https://github.com/AuroraWright/Luma3DS/blob/master/k11_extension/Makefile#L17 https://github.com/AuroraWright/Luma3DS/blob/master/Makefile#L56