LumaTeam/Luma3DS #1092: Majora's Mask 3D Crashing at First Loading Screen

hbz1, 2018-06-05:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

9.0-2a840f2

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: (x)

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Once opening Majora's Mask right after the 3DS splash screen, the game loads for half a second then crashes. Running the Japanese version. Tried changing the region using Luma Locale Switcher and no hope. Changing the language on the US version doesn't work either.

**Steps to reproduce:**

1. Download JP Majora's Mask through freeShop.

2. Launch game.
[image]

**Dump file:**

Hikari-chin, 2018-06-05:
We don't provide support for pirated software here.

closed 2018-06-05 by Hikari-chin

Darkitty, 2018-06-25:
Hello,

I get the same with a real EU cartridge with my 2DS LL. But, it's work perfectly on my 3DS LL. I tried with two different 2DS LL with the same issue.

Thanks for Luma3DS ❤️