LumaTeam/Luma3DS #1093: When booted from NAND, LayeredFS patches are not loaded from SD

thejsa, 2018-06-09:
**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39E

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

9.0 stable

**Luma3DS configuration/options:**

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

--

**Explanation of the issue:**
In sysmodules/loader/source/patcher.c , the LayeredFS functions call `openLumaFile()` to load files. However, when Luma3DS is booted from the NAND, this leads to the LayeredFS code looking for patch files on the NAND, which is incorrect behaviour.

**Steps to reproduce:**

1. Install Luma3DS to CTRNAND

2. Remove boot.firm from the SD card

3. Enable game patching

4. Copy a game patch to the SD card -- it will not be loaded.

Hikari-chin, 2018-06-09:
This has been reported before and it was determined that it was *not* incorrect behavior, but rather intended. Even, booting from NAND is often called "booting SDless".

closed 2018-06-09 by Hikari-chin

thejsa, 2018-06-09:
The wiki page states that patch files should be placed on the SD card - this is incorrect? https://github.com/AuroraWright/Luma3DS/wiki/Optional-features

Hikari-chin, 2018-06-09:
No; it's just that boot.firm should *also* be placed on the SD card, if available. NAND booting is intended to be for situations where the SD card is not available.

thejsa, 2018-06-09:
Ah, that makes sense. Alrighty then