LumaTeam/Luma3DS #1094: Arm11 crashing all the time on clean luma install

kizoLeonardoMontiel, 2018-06-10:
System model:
Old 2ds

SysNAND version (+emu/redNAND version if applicable):
11.6.0-39U SysNAND

Entrypoint (How/what you're using to boot Luma3DS):

Sighax

Luma3DS version:

Luma 3ds 8.1.1

Luma3DS configuration/options:

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off )

PIN lock: ( off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

Explanation of the issue:

Whenever I try to load some specific games such as Pokémon Rumble Blast (USA) I get an ARM11 crash with the prefetch abort (svcBreak) error in it. This also happens at random at multiple other games such as Kirby Triple Deluxe and Pokémon Ultra Moon, although I haven't been able to reproduce the issue yet as I've just done a clean CFW install to no avail. 
Old install was boot9Strap + luma 8 and luma 9, while the new install is sighax with luma 9 
Both installs had the same issue.

Steps to reproduce:

1. Load Pokémon Rumble Blast or play Pokémon Ultra Moon/Kirby Triple Deluxe for extensive periods of time

2. Consistent crashing on Pokémon Rumble Blast's first loading screen, while on the other games it crashes at random, see picture below:
[image]

Dump file:
[arm11.zip](https://github.com/AuroraWright/Luma3DS/files/2088052/arm11.zip)

WubTheGame, 2018-06-11:
Sighax? Yeah, I am pretty sure Sighax wasn't meant for the end user. Boot9Strap is for that.

Hikari-chin, 2018-06-11:
Can't reproduce this, but errors like these are usually from SD card failures (corruption, hardware damage, etc.) or from getting your games illegitimately from unscrupulous sources.

closed 2018-06-11 by Hikari-chin