LumaTeam/Luma3DS #1095: Activating soft-reset combo in DSiWare WarioWare Touched causes prefetch abort

Robz8, 2018-06-11:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

9.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When activating the soft-reset combo in the DSiWare version of WarioWare Touched, or just doing the soft-reset method in TWL mode in general, I get an ARM9 prefetch abort.

This has never happened before, but after trying an unofficial Luma build, this issue occurs. Even switching back to latest official Luma doesn't fix it.

**Steps to reproduce:**

1. Open DSiWare version of WarioWare Touched.

2. Hold L+R+START+SELECT to activate soft-reset.

**Dump file:**
[crash_dump_00000005.zip](https://github.com/AuroraWright/Luma3DS/files/2088574/crash_dump_00000005.zip)
[image]

Robz8, 2018-06-25:
Yes.

MojoJojoDojo, 2018-06-29:
This is true for every game you try to play on nds-bootstrap

ImmortalScientist, 2018-07-02:
Can confirm this affects any game played through nds-bootstrap. Both key combinations (R+L+St+Se and R+L+Down+B) have this effect.

Robz8, 2018-07-02:
One can avoid this issue by using Rei-Six, instead of Luma.
Can confirm this on my side.

TuxSH, 2018-08-01:
@Robz8 0xc8804465 is "mount point not registered". If this is what I think (a regression (?)), the issue shouldn't occur if luma is booted from nand.

piratesephiroth, 2018-08-03:
@TuxSH yep that's it.

closed 2018-08-21 by AuroraWright