LumaTeam/Luma3DS #1096: Aurora Wright I need some help, I hope it is a easy fix.

ratmandom, 2018-06-22:
NOOBIE DOM

[Doms crash dump.zip](https://github.com/AuroraWright/Luma3DS/files/2129210/Doms.crash.dump.zip)

-- Please fill in the placeholders.-->
 *System model:*

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

In system settings it says / Sys 11.6.0-39E /

**Entrypoint (How/what you're using to boot Luma3DS):**

it says Booted from sd via B9S

**Luma3DS version:**
LUMA version 9.0-2a840f2
**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: (4 )

Splash: ( off)

PIN lock: (off )

New 3DS CPU: (n/a )

--

Autoboot EmuNAND: (off)

Use EmuNAND FIRM if booting with R: (off )

Enable loading external FIRMs and modules: (just done my 3ds last weekend )

Enable game patching: ( off)

Show NAND or user string in System Settings: (on )

Show GBA boot screen in patched AGB_FIRM: ( off)

Patch ARM9 access: ( ?)

Set developer UNITINFO: (off )

Disable ARM11 exception handlers: (off )

--

**Explanation of the issue:**
when I try to run a 3ds game some will boot of the sd card others will come up with this error
homebrew as well a lot will not work
I have tried checking the sd card for errors no problem there

hey if you can help me out I have messaged you on GBAtemp,
my email as well is ratmandom@outlook.com

I need to learn this stuff my self. in my family we have professors and doctors, a few people who are good with computers. my great granddad help make one of the first computers in world war 2 just good people in general.

I shine TBH more than most people in this family
and I only started with this stuff with a wii u about 2 years a go and I am 24 now.

sorry for the English I have a lot of learning disability's, are you Italian if so I am pure from 3 generations a go.

I would like to say hello as well on gbatemp

name=ratmandom
rat=from the pet

**Steps to reproduce:**

1.

2.

**Dump file:**

ratmandom, 2018-06-22:
yeah it gets better I am autistic, dyslexic with ADD, schizophrenia and bipolar all 5 at the same time.
to be honest I am not the average person or computer person, like I do kickboxing and listen to trap music and loads of people don't have a clue what I do. I am just waiting for some thing to happen. but far from depressed or violent in the bad ways. why not be this way? you probably know at least some of what I am saying...

yuukieve, 2018-06-22:
Not a Luma issue, rather an issue of a corrupted or bad SD card.

closed 2018-06-22 by yuukieve

ratmandom, 2018-06-22:
thank you yuujieve
I will test out my old 4gb sd card

the card I was using was a 256gb sandisk card and I did test it for errors with h2testw no problems there.
I should look up a recommendation then of good SD cards for the 3ds.

feww it is not my 3ds !
dam sd card is not a micro(not good for my hardware) lol

yuukieve, 2018-06-22:
If h2testw reported no problems it may not be a direct hardware issue with the SD card, but rather corrupted data on the card itself.

ratmandom, 2018-06-23:
should I send my sd card back, it is from amazon so I could very easily or could I fix this my self if I can.

pbanj, 2018-06-23:
the issue might just be the size of the card. bigger sd cards cause a slowdown. lots of people tend to have issues with even 128GB. try reformatting the card but this time use 64kb for the cluster size instead of 32 and see if the issue still happens

ratmandom, 2018-06-23:
ok I have 3 more dumps if you could just check them, if there is not a fix I need a new 2ds xl.
[Doms dump test 1.zip](https://github.com/AuroraWright/Luma3DS/files/2129910/Doms.dump.test.1.zip)

ratmandom, 2018-06-23:
can you just check these dumps out if they do say the same thing.
I need a new 2ds xl.

[Doms dump test 1.zip](https://github.com/AuroraWright/Luma3DS/files/2129912/Doms.dump.test.1.zip)
[Doms dump test 2.zip](https://github.com/AuroraWright/Luma3DS/files/2129913/Doms.dump.test.2.zip)
[Doms dump test 3.zip](https://github.com/AuroraWright/Luma3DS/files/2129914/Doms.dump.test.3.zip)