LumaTeam/Luma3DS #1099: Black screen on boot (awaiting home menu loading)

lorilith, 2018-06-25:
**System model:**
All models

**SysNAND version (+emu/redNAND version if applicable):**

11.6.0-39[All] through 11.7.0-40[All]

**Entrypoint (How/what you're using to boot Luma3DS):**
b9s 1.2/1.3

**Luma3DS version:**
9.0

**Luma3DS configuration/options:**

Show NAND or user string in System Settings: ( X )

all others off

**Explanation of the issue:**

System occasionally will hang on home menu load.
Black screen. Rosalina menu can be accessed.
GM9/firmware payloads/chainloader is accessible on boot
Luma config is accessible on boot

clearing home menu extdata does not fix issue (nor does removing SD card or gamecart)

**Steps to reproduce:**

random occurrance. usually only happens once per 3ds/2ds. can happen after initial install of b9s/luma. can also happen on force power off during assumed hang/freeze.

fix action usually requires user to wait 5-10 minutes (in my experience it is closer to 5-6 minutes) and then the home menu will load.

repeated attempts to reboot does not fix this issue until user waits the 5-10 minutes.

Issue is commonly seen on the NH discord. attaching an occurrence from tonight.
[image]

sergioberns, 2018-06-25:
I had the exact same problem here. I tried some of the procedures published at 3ds.hacks.guide, but the only solution was that you offered in this thread (leting it sit on the black screen for about 10 minutes).

After that, my 3DS is working just fine. Thanks for the support!

lorilith, 2018-06-26:
[image]
Another instance from tonight

lorilith, 2018-06-27:
another from tonight
[image]
[image]

lorilith, 2018-06-27:
not sure what kind of metric we need for this, going to stop posting occurrences as i believe these examples are evidence of an existing intermittent issue.

fincs, 2018-06-27:
I'd like to ask a few questions.

- How many 'apps' do users affected by this have installed on their screen?
- What kinds of SD cards are affected users using?
- Have they been tested for errors?
- What's the system history of these 3DSes? Has anything else been previously installed on them, like older exploits or tools?

lorilith, 2018-06-28:
i have seen this on a fresh install with no apps other than sudoku and system apps using the stock 4gb sd.
not sure about testing for errors.
new systems as well as old with many different histories

profi200, 2018-09-04:
Never had this problem on any 3DS.
May be related: There is a problem with N3DS consoles when the camera is defective(?). Such systems will not show HOME menu at all but there is sound. Recovery mode works and all homebrew FIRMs do too. Disconnecting the cameras fixes it.

MarioMasta64, 2018-09-09:
i too experience this, i have ~70 titles from eshop atm (and for not working camera's you can uncomment some code in the luma source and compile it @profi200 )

zipplet, 2018-09-11:
Luma 9.1 here on N2DS XL. Maybe this is triggered by StreetPass in sleep mode.

My system had been OK for days (sleep mode being used regularly) but today I received a streetpass hit while in sleep while on the main menu. Indicator LED was green.

Opened the display and the indicator LED turned off, power LED on, wi-fi LED blinking the usual pattern it does when wi-fi is enabled on home menu but more slowly than usual. Black screens.

Powered off. Powered on. Black screens. Had to leave the system for 5 minutes for it to recover by itself.

euandersson, 2018-10-26:
Hi Guys, I had this problem for the first time yesterday, and I could see that the problem is in the memory card, maybe in the slot, maybe in the micro SD adapter, I left it in that mode you said, 1h and nothing he did not return. The night when I came from work I did the following, I called him without the SDcard, ok he called normal, then I inserted the SD and I pressed it for a few seconds and voila, the games appeared, then I hung up and called again and everything was normal. I have a R4card 2015 and sometimes I have to hold it in this tight way to the bottom so I can just drop it, so I decided to try the same with the 3ds SDcard.

Have a good day / night guys.

**old3DS 32GB Firm 11.6 + Luma 9.1**

MechanicalDragon0687, 2019-06-02:
> Hi Guys, I had this problem for the first time yesterday, and I could see that the problem is in the memory card, maybe in the slot, maybe in the micro SD adapter, I left it in that mode you said, [...]

this is a different issue and is unrelated. it looks the same to you but it isnt the same issue.

ParzivalWolfram, 2019-06-02:
Multiple 2DSes i've hacked had this issue. Found the following to work: start console, then immediately flip sleep switch ON, wait for network indicator to slowly blink, then flip sleep switch OFF. The HOME Menu will load. This only starts on the second boot after initial Luma install, and after this procedure, never happens again.

daemongloom, 2019-07-16:
Got the same issue on new 2ds xl. LumaRDS v10.0.1, latest official euro firmware. Shutdown/poweron and reboot does not fix this issue. Home menu appeared after waiting for a few minutes on black screen.

pthfdr-42, 2019-09-05:
It only happens to me when I have a flashcart (DSTTi and AK2i in my case).
If I remove the flashcart, it would boot as usual.

Michael18751, 2019-09-25:
Had this issue but removed the game cart and it recovered

Lexxtalion, 2020-02-26:
Hi Everyone, i had issue with my n3ds, when i boot up with or without sdcard it just black screen and just give theme sound then silent, but when i press any button n3ds will reboot then blackscreen and sound again. but i can access my luma or godmode. i already try to wait until 30 minutes nothing boot up. Please help me, i want play some game on my n3ds again .

ParzivalWolfram, 2020-02-26:
That's a bit off-topic for this issue, but it sounds like you're getting an
error handler when booting. Rename the "sdmc:/luma/dumps" folder (or
whichever crashdumps go into, idr the name at the moment) and press A twice
when it would normally reboot. Does that generate a dump?

shawnz, 2020-03-22:
I also had this issue on an old 3DS, only after enabling streetpass and going out in public, and I was able to resolve it by turning off the wireless switch as some others have indicated. Luma 10.0.1, FW 11.13.0-45U

1031982, 2020-05-08:
Just for another opinion, I just updated to 10.1.3 using the updater, and I had the same issue. It wouldn't matter if I was starting with or without the SD card inserted.
I fixed it by using the previous version, 10.1.2. After I knew everything was working, I updated the old fashion way. I tested by rebooting my system multiple times, and have not had an issue.

zipplet, 2020-05-09:
Silly question maybe... has anyone tried dumping the filesystem of the NAND when this happens, then comparing it to a dump taken after it is allowed to sit and cure itself to find out why?

raspimana, 2020-05-15:
[This works for me](https://www.ifixit.com/Answers/View/422753/3DS+Black+Screen+When+Booting+Up-+Has+CFW#answer422905)

1031982, 2020-05-15:
> 
> 
> [This works for me](https://www.ifixit.com/Answers/View/422753/3DS+Black+Screen+When+Booting+Up-+Has+CFW#answer422905)
> 
> EDIT: Should still fix that bug.

That only works if you don't have a New 2DSXL. Otherwise, you are looking at getting a tri-wing screwdriver, and doing this with the bottom case off and the speakers disconnected.

raspimana, 2020-05-15:
> That only works if you don't have a New 2DSXL. Otherwise, you are looking at getting a tri-wing screwdriver, and doing this with the bottom case off and the speakers disconnected.

Yeah, a little more expensive than just patching out that bug.
(By expensive I mean you need to get a tri-wing screwdriver if you don't have one)

Sabre8073, 2020-05-20:
I'm having a similar issue. I experience a black screen that I believe is activated during sleep mode and can only be resolved by a hard reset. Recently started occurring within the last official update and luma updates.

raspimana, 2020-05-21:
Happens to me too. I can’t remember what app it was on. I think it was on GM9.

ParzivalWolfram, 2020-05-21:
Interesting that someone had sleep mode trigger it, sleep mode is the
solution in 90% of my cases.

AuroraWright, 2020-11-17:
Possibly https://gbatemp.net/threads/black-screen-at-the-start-of-the-3ds.543980/page-3#post-8763629 aka a home menu issue with corrupted sysdata/extdata

ParzivalWolfram, 2020-11-17:
It's not always corrupted, though, as a restart plus like the sleep switch
being held/flipped or smth under some circumstances fixes it, which
wouldn't be true if it was a corrupt home menu, which would require a hell
of a lot more work to fix.

AuroraWright, 2020-11-17:
It's still a home menu issue most likely, Luma only applies very minor patches to the home menu (specifically to patch out the region checks and the like) so it would happen on stock too imo.

furqaan44, 2021-03-11:
I need help for my old 3ds xl 
Please i try everything but nothing work 
My 3ds is off and when i turn it on it show tge blue light and turn off again

furqaan44, 2021-03-11:
When i turn on the 3ds withoute a fat32 sd card it remain with the blue light and when i put an sd card in it,it turn off

furqaan44, 2021-03-11:
I try replacing boot.firm but nothing work

furqaan44, 2021-03-11:
Please help me i dont want to lose my 3ds

tbvns2, 2022-07-06:
I actually have this issue rn 

Do I just need to wait for it to be fixed ?

impeeza, 2022-07-06:
No. You need to read the posts and read the answer on https://github.com/LumaTeam/Luma3DS/issues/1555

tbvns2, 2022-07-06:
nope i just waited and it worked

tbvns2, 2022-07-06:
it just took 07:43.802

stckgrnmnstr, 2022-09-25:
on ofw v11.5 if you have luma v9.1 and Godmode v2.1; you can restart the 3ds by using your boot.firm.bak file and turn it into a boot.firm file by erasing the .bak and startup Godmode(with start button). You need to update Godmode on the SD card and then run the 2.6 update into the NAND and it works as a small "bypass"; so now every time you start your 3ds it will either boot normal or boot straight through Godmode 2.6. This works after the soft-bricking effect that the ofw gave during the 11.5 update; so if the console won't turn on or it has a black screen or a flashing blue led; this gets it back up and running. I haven't tried to update to any 11.6 ofw but mines still on Luma v9.1 and Godmode v2.6.

KingOfShit112, 2023-02-24:
can someone help me :(
I just starting with this problem where i cant normal Boot my old 3ds i tried to update the luma version but aint working and i tried to wait the 10 minutes but also aint working, i'm using the luma v12.0. i dont really know what i'm a doing haha but a dont know if anyone have a solution for this problem.
I read that playing super mario bros 2 golden edition may cause this problem but i don't know if thats the reason.
I see that if i use L+power it start but if i use the home buton it send me to the luma3ds configuration.
is there any solution or is better if i use the v9.5 of luma?
it seems to be freez but i don't really know about this.

OutrageOrange, 2023-04-16:
This appears to be a gamecart slot compatibility issue. I opened gm9 because it was stuck on black screen, I opened the gamecart directory and the system crashed.

I removed the game in the slot, switched the power on, and immediately put the system to sleep and it started working again.

megDreyer, 2024-09-08:
It seems as though this issue still exists on 13.1.2, old 3DS XL. Holding select on power on gets me to the Luma3DS configuration menu. From there, I can select boot chainloader. I can get into godmode 9 with no issues, but I am unable to boot into the home menu from either the chainloader or a regular power off. Blue power light stays lit, and if I hold the power button it'll turn off after a while.

stckgrnmnstr, 2024-09-08:
> It seems as though this issue still exists on 13.1.2, old 3DS XL. Holding select on power on gets me to the Luma3DS configuration menu. From there, I can select boot chainloader. I can get into [...]

I had to do a boot.bak to turn it on, load into Godmode9 and flash a Godmode9 update onto the 3ds hardware with the boot.bak being treated as the boot.firm and keep luma 9.1

megDreyer, 2024-09-11:
Just circling back to say that it has fixed itself and now boots regularly without issue, but is unable to launch any title. I will make a separate bug report for that issue.

megDreyer, 2025-03-02:
> Just circling back to say that it has fixed itself and now boots regularly without issue, but is unable to launch any title. I will make a separate bug report for that issue.

So. I appeared to have two separate issues that I was confusing as one. They are detailed as below. One was a hardware issue and my fault.

Condition 1: Able to boot into home screen, but when opening any title, the device hard freezes. This was an issue with my wifi card not having been seated properly.
Fix: seat the wifi card properly.

Condition 2: 3DS was left in my bag, powered on, with a game cart in and that game launched. The game cart got bumped and popped out. I pushed it back in. Black screens on top and bottom, but able to launch the config menu by holding start. It would not boot into home menu. I was able to power off by holding power button. 
Fix: power on by pressing power button. Leave the device open on my table for a few minutes (it took about 4) and it booted into the home menu without difficulty.
I'll try and reproduce the crash upon game cart removal issue.

samiam567, 2025-03-22:
> I had the exact same problem here. I tried some of the procedures published at 3ds.hacks.guide, but the only solution was that that you offered in this thread (letting it sit on the black screen for about 10 minutes).
> 
> After that, my 3DS is working just fine. Thanks for the support!

This worked for me as well!

Siborgatron, 2025-04-06:
Removed the game cartridge & it booted. Something must have broken inside from the cartridge being in there for months. Now it's not reading the 2DS cartridge that was in there & only sometimes reading 3DS cartridges.

(Had taken the 2DS apart to see if it was the Wifi detached, wasn't the same model as instructional videos. Then made a NAND backup, freaking out because I hadn't already made one in case it was a software issue.)

mirusu400, 2026-09-05:
I also struggling with this issue, (I almost turned on for 24 hours but it keeps black screen) and here are two solution:

1. faulty wifi board -> Try https://wiki.hacks.guide/wiki/3DS:Ctrcheck and use this luma fork: https://wiki.hacks.guide/wiki/Special:DownloadWithFilename/File:Luma_ignore_cfgnor.firm/boot.firm but finally you need to change wifi board

2. wrong rtc clock -> https://wiki.hacks.guide/wiki/3DS:RTC_black_screen_bug