LumaTeam/Luma3DS #1101: Blue light filters!

joo456, 2018-07-11:
Blue light is a problem with most devices where at night the light from screens causes the brain to be tricked into thinking it is daytime, therefore keeping the user of the device awake for much longer. There are programs that aim to fix this problem, such as [f.lux for Windows, Mac, jailbroken iOS devices and Android devices](https://justgetflux.com/) and [redshift for Linux](http://jonls.dk/redshift/). However, there is a 3ds recreation of redshift called [CTR_Redshift by MarcusD.](https://gbatemp.net/threads/ctr_redshift-hardware-based-blue-light-filter-for-old3ds-and-2ds.493736/) The application works quite well (but is quite annoying to set up using the app) and could (i think) be ported easily to the rosalina menu or as an option in luma config. There is also a [pull request adding blue light filters](#1065).

Robz8, 2018-07-11:
There's an unofficial Luma build that has Redshift ported into Rosalina, also made by MarcusD.
https://puu.sh/AxWVU/a6635739b3.firm

joo456, 2018-07-11:
nice! thanks robz

therealbungus, 2018-07-12:
Watch what you say about the PR otherwise your comments will get deleted

Thanks AW

yuukieve, 2018-07-12:
There's already an open PR for this, as you said.

closed 2018-07-12 by yuukieve