LumaTeam/Luma3DS #1106: LUMA 3DS CONFIGURATION STAYS BRICK

mxnumolina, 2018-07-24:
Hi, I know this is not for support but in the discord they can not help me. I have been trying to install the Luma and just in the configuration I press start and nothing happens, it gets caught. Does anyone know what happens? I have everything updated, and use the SeedMiner method.
My version of the System is: 11.7.0-40E

rctgamer3, 2018-07-24:
>I know this is not for support

.. Then, don't post here?

rctgamer3, 2018-07-24:
Ask over at GBATemp.

mxnumolina, 2018-07-24:
Yes, I have published, but it has been because in Discord they can not help me.

Jwren76, 2018-07-25:
I am having this problem as well. After updating Luma to 9.0, bootstrap to 1.3 and Sys to 11.7.0-40E my new 3DS XL with only boot to the Luma Configuration screen.  Pressing start to exit causes hang and I can only power off by holding power button for 15-20 seconds.  I can boot to home menu by holding L while powering on.  I've searched for weeks now with no solutions.

Jwren76, 2018-07-25:
Also I'm having weird issues where the home button doesn't work in Godmode9 or if I want to return to home menu after pressing power button. Not sure where I went wrong in the update.  Happened immediately after trying to update Luma, B9S, and FW following 3DS Hacks Guide.

yuukieve, 2018-07-25:
Can't reproduce this behavior, even with an emuNAND (per the "booting with L" works stuff).

closed 2018-07-25 by yuukieve