LumaTeam/Luma3DS #1107: 11.8.0 Critical Issue. Luma 3DS Patched. Luma Update Required.

Merik2013, 2018-07-31:
I'm sure many of you working on the CFW already know this or soon will by the time you're reading this, but I figured someone needed to post this in Issues regardless. I don't see much need to fill this in too with much detail as the details in this case aren't available to a user such as myself and the problem affects ALL versions of Luma3DS.

I will add this though for reference. This update did NOT patch B9S (I doubt Nintendo really has a way to do that to begin with), just Luma3DS. As a result, if someone were to update to 11.8.0 they just need to follow these instructions to boot into the Native Firm while we wait for a solution to this issue on your end.

    1. Download the appropriate NATIVE_FIRM for your device here (https://mega.nz/#F!uNhHQTCJ!EEndzd7lfv97skLBTUelRw).
   2.  Rename the file you downloaded to native.firm and copy it to /luma/ on your SD card.
    3. Reboot your device holding SELECT; this will open the Luma3DS configuration menu.
    4. Scroll down to Enable loading external FIRMs and modules and enable it.
        (It should look like: (x) Enable loading external FIRMs and modules.)
    5. Press START to save the new Luma3DS' configuration settings and reboot the device.

AuroraWright, 2018-07-31:
It's not a critical issue and it's fixed in https://github.com/AuroraWright/Luma3DS/commit/89cc6ede1a5dc3f846cf508c5f698a869e9e1018

closed 2018-07-31 by AuroraWright

Shadow322, 2018-07-31:
@AuroraWright can u push it to a compiled release for everyone?

mysterypaint, 2018-07-31:
Thank you for the quick update! - hazel

duranlisandro, 2018-07-31:
Nice! Thank you so much!

thisisloze, 2018-07-31:
Thanks! Waiting on compile!

AuroraWright, 2018-07-31:
[boot.firm.zip](https://github.com/AuroraWright/Luma3DS/files/2243657/boot.firm.zip)

llanos1205, 2018-07-31:
i cant turn it off

jvhellraiser, 2018-07-31:
turn off for me o3ds xl

Shadow322, 2018-07-31:
@AuroraWright um that boot.firm isnt the new one

Shadow322, 2018-07-31:
@AuroraWright that noot.firm isnt latest commit its the one right before dd6e071ddead5dacdcf3e8be7a0671f4392c4b9f <-before latest

AuroraWright, 2018-07-31:
It's latest, I compiled before I committed

Shadow322, 2018-07-31:
oh ok so rosalina and luma config shows wrong hash

greninjaman1998, 2018-07-31:
How do i get the latest native

Please help

ShadowOne333, 2018-08-01:
Can we use the latest 9.1 Luma on a 3DS which still has 11.7?
Are there any cons to doing this? And if so how would one go about using 9.1 with non 11.8 consoles?

For a 11.8 console, do we only need the Luma 9.1 boot.firm or do we need something else?