LumaTeam/Luma3DS #1108: HALP

disasterinlove, 2018-07-31:
I accidentally downloaded this onto my ds hardware itself.  I have take out my memory card and it is not on there but still will come up when I press select.  I need to know how to take this off my ds as it is not working now with the new update.  I never even meant to download this but apparently I did.  How do I remove now when my ds isnt even working!?

yuukieve, 2018-07-31:
Not a tech support forum.

closed 2018-07-31 by yuukieve

disasterinlove, 2018-07-31:
Can you link me to where I can get there?

disasterinlove, 2018-07-31:
Please?  Im literally over here in hysterics because Ive lost everything.

ihaveamac, 2018-07-31:
#1107

yuukieve, 2018-07-31:
If you have trouble figuring out what to do from #1107, try the [Nintendo Homebrew Discord, as linked in the ISSUE_TEMPLATE.md](https://discord.gg/C29hYvh)