LumaTeam/Luma3DS #1110: [Not an issue] Make a new (dev or nightly) branch = Avoid nightly builds for patchs

GGLinnk, 2018-07-31:
[Not an issue]
Hello,

This recent situation point a flaw about Luma3DS development, there is only one branch where stable and nightly sources are saved, causing difficulties to quickly add an hotfix if needed.

What I recommend is to create two branches : stable and nightly.
And if you are courageous, a branch by function.
So, when you want to add a new function, just have to merge branches,
If patchs are needed, do it without nightly issue.

PS : I'm currently learning english, so, if there is any lack or confusion, please, say it to me.

GGLinnk, 2018-08-01:
I just make the stable branch, i will pull-it in few mins.

yuukieve, 2018-08-01:
This is not really necessary; very rarely are nightlies non-functional, and very rarely does a system update break Luma; 11.8 was an exception rather than the rule.

closed 2018-08-01 by yuukieve