LumaTeam/Luma3DS #1111: Luma Installer

StatusQuo1985, 2018-08-01:
This is not an issue, it is more of a request to distribute your work.

I have compiled a program called "Luma Installer" designed to take all the head work out of installing Luma via the NtrBoot method, to make it even more noob-proof.

https://gbatemp.net/download/luma-installer-public-release-ntrboot.35037/

This installer is released as a package containing multiple versions of Luma. Now, as you give Luma away for free, I cannot see this being an issue, but figured I should at least ask, as I am distributing your work.

Please feel free to download and examine my work, I am not a particularly skilled programmer myself, but the tool works flawlessly and I use it frequently.

noirscape, 2018-08-01:
Luma3DS is released under the GPLv3, information of which you can find
here: https://choosealicense.com/licenses/gpl-3.0/

Looking at a quick glance, you'd need to link back to the source code on
your release page and provide the source code for your own installer, as
well as distribute your installer under the same license.

In addition, for the bundled tools, you'd also need to apply their licenses
(MIT and the GPL), which means linking to these projects as well as well as
providing proper credit.

Note: IANAL.

In addition, I'd advise getting rid of the piracy tools and CIA game dump
you included (why?) as this goes against the rules of GBATemp as a site and
I'm pretty sure against the rules on this issue tracker as well.

StatusQuo1985, 2018-08-03:
Ok, cool. Thank-you for that information. I will make the changes as soon as I get a chance to actually sit down and do it

yuukieve, 2018-08-31:
Closed pending further information/activity from the poster.

closed 2018-08-31 by yuukieve