LumaTeam/Luma3DS #1112: Encountered Crash Dump screen, followed by a Brick

PP-theSLAYER, 2018-08-01:
Sadly, I'm unsure what version of Luma I was on, but I haven't updated to the newest Luma and didn't update to newest 2DS firmware. (I was probably on the previous FW version, since I was able to use Pokebank within the past month or so)

As a matter of fact, I was deleting titles using FBI before the brick, in preparation of the updates I was gonna do.
I know I didn't screw up title deletions, as I rebooted my 3DS a few times, before said BSOD.
If I'm not mistaken, I was exiting Rosalina (from inside Health and Safety).

Crash screen popped up, then 3DS was blue screen on reboot.
Attached to this message, is the latest Crash Dump I see in the Luma folder.
[crash_dump_00000004.zip](https://github.com/AuroraWright/Luma3DS/files/2249561/crash_dump_00000004.zip)

Text on the BSOD:
00F800FE
00000000 00000000
00000200 00000000

I'm kinda surprised that my 2DS managed to get bricked on B9S.
Ultimate, I'm not here to point fingers; bricks are a risk of using CFWs so I was mentally prepared for it.
Just want to prevent it from happening to others. I'll try to provide any other information, if needed.

Multimegamander, 2018-08-01:
Do u have a NAND Dump, also, use fastboot3ds

TuxSH, 2018-08-01:
What you use (b9s) is unrelated to the issue.

The crash dump is fs panicking.

00F800FE 00000000 00000000 00000200 00000000 is a "Error when having SD-card reader connected to NAND during boot"

This looks like a hardware issue to me, not a software one. Do or did you have a hardmod? If you can, could you check the motherboard?

PP-theSLAYER, 2018-08-01:
Thanks for the suggestion Multimegamander.

Based on the files on my 3DS's SD card, I have a NAND dump on it. (_NANDmin_sys.bin_)
(I have a habit of storing and labeling all my nand dumps in my external hard drive, but I'm overseas for the time being, so I don't have access to that external drive)

Additionally, I can't use fastboot3ds right now; NTRbootflasher claims that my Gateway Blue card is incompatible with ntrboot flashing, so I need to get a different NDS flash cart, in order to use ntrboothax to restore my NAND.

To add on, it looks like I'm not the only one affected by this issue.
I've read another one whether the use got bricked, the only difference is his/her's isn't BSOD,
potentially due the force shutting down of the system after a glitched screen appeared, when leaving Rosalina.

https://github.com/AuroraWright/Luma3DS/issues/1098

PP-theSLAYER, 2018-08-01:
Hey TuxSH, I have a hard mod (which I didn't want done but the guy did it. It was one of those guys that ghosted people for a long time on gbatemp, so I was just happy to get my console back), but I've never used it.
Plus, the cable-SD dongle never worked on my laptop, so I didn't bring it overseas (was afraid the customs would think it looks suspicious)

What specifically you want me to look for on the motherboard? It's a clear unit, so I easily look through the shell-housing

TuxSH, 2018-08-01:
@PP-theSLAYER you won't be able to restore your nand because your nand is not correctly connected in the first place.

This is **not** a software issue, @Multimegamander suggestion is _not_ helpful.

PP-theSLAYER, 2018-08-01:
You're saying my Nand was dislodged from the motherboard or something?

TuxSH, 2018-08-01:
Oh hm thanks for the quick reply.

1) do the other person(s) who has/have this issue have had their hardmod done by the same person?

2) I think you should check these points: https://www.reboot.ms/forum/virtualdr/www.virtualdrive.cc/2DS_NAND_pinout.jpg , I'm asking more knowledgable ppl for that in the meanwhile.

> You're saying my Nand was dislodged from the motherboard or something?

Probably a contact failure or something like that. Currently the bootrom think you've connected the sd card bus to the nand pinouts (or something like that)

PP-theSLAYER, 2018-08-01:
I've got no clue. My 2DS has been functional for at least a year since I've got it back, so never thought about it.

I probably got to take my 2DS apart, in other to see those points.

PP-theSLAYER, 2018-08-02:
@TuxSH here are two shots of the cables on the motherboard.

If it's really the hardmod, would cutting off the wires solve the problem?

Also, there's a disconnected black wire. Not sure if it was disconnected during the opening, tho.
[image]
[image]

pbanj, 2018-08-02:
ok so first let me say sorry it took me so long to reply here. my kid has been sick and yesterday was hectic. 

ok so you can go about solving this one of two ways. one you can try adding a little tape over the wires/points so they dont bridge against the screen. that black wire is ground for the hardmod, it being off means the hardmod wont work anymore. but it shouldnt cause an issue. from looking at your pics it looks as 2 wires got pinched and is most likely the cause of the issue as it looks like bare wires. so you can try and tape it like i said or snip all the wires of and out of the system. if you do that i would still add a little tape over the points just to be safe.

idk if you use discord but if you want to talk more about it you can reach me here @pbanj#9188

PP-theSLAYER, 2018-08-02:
hey @pbanj , I'm sorry to hear that your kid is sick. Hope your kid gets well soon!

Thanks for the reply, I've already jumped the gun and cut off all the wires, since another one broke between the sessions of opening and closing the lid.

After I cut off the wires, the issue returned, and I realized that the back of the screen was partially conductive, so the small remnants of solder probably made it BSOD again. I taped the back of the screen, and everything peachy now! (tho I destroyed my rear camera ribbon by accident, but those are cheap on ebay, so not too concerned)

Anyhow, thanks for the advice, if I didn't already jump the gun, your advice would definitely been very useful in helping me figure out how to solve the calamity.

With that, this thoroughly proves that Luma isn't the issue here, as such I'm going to close this ><

closed 2018-08-02 by PP-theSLAYER