LumaTeam/Luma3DS #1114: ntr cfw broken on luma 9.1

GamingMaster777, 2018-08-02:
after updating from 9.0 to 9.1, ntr cfw would crash any game before the 3ds splash screen is finished. after restoring back to 9.0, ntr cfw and cheats work once again. i currently am on 11.7 as well

CodyMKW, 2018-08-02:
https://github.com/Pablomf6/BootNTR/releases/tag/v2.12 download this version

GamingMaster777, 2018-08-02:
k thx

closed 2018-08-02 by fincs

PatrickB1995, 2018-08-05:
I am STILL having problems. I am running Luma 9.1 on Firmware 11.6.39U on the New 3DS XL and am using the latest PabloMK6 NTR as listed above with the latest ntr.bin file Every time I launch NTR CFW, I get this:

#FILEOPEN_FAILURE 
#LOAD_AND_EXECUTE

Press Any Key To Reboot

I have tried uninstalling and reinstalling, checked my SD card for any old NTR files, etc and I am at a loss...

What the hell did I do wrong? I have spent over 3 hours now trying to get NTR to run and its been a MASSIVE headache...

PatrickB1995, 2018-08-05:
Also, I have tried all the other latest NTR released versions that released a day ago and zero luck there either...

PatrickB1995, 2018-08-05:
I am simply just trying to ask for any ideas or solutions for fixing this. I dont know where to ask... I would really appreciate some help.

Shadow322, 2018-08-05:
Like said THIS IS NOT A SUPPORT FORUM

PatrickB1995, 2018-08-05:
Then instead of BEING A CUCK, you can point me in the right direction of where I can ask for proper help INSTEAD OF USING CAPS LOCK

Thanks!

ihaveamac, 2018-08-05:
sigh