LumaTeam/Luma3DS #1118: Luma3DS configuration menu/chainloader can be brought up when launching the DS Connection Settings menu

zacchi4k, 2018-08-10:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.7.0-40E

**Entrypoint (How/what you're using to boot Luma3DS):**
Fastboot3DS 1.1 beta

**Luma3DS version:**
9.1 stable

**Luma3DS configuration/options:**
Default EmuNAND: (N/A)
Screen brightness: (3)
Splash: (None)
PIN lock: (None)
New 3DS CPU: (None)
Autoboot EmuNAND: (N/A)
Use EmuNAND FIRM if booting with R: (N/A)
Enable loading external FIRMs and modules: (No)
Enable game patching: (Yes)
Show NAND or user string in System Settings: (Yes)
Show GBA boot screen in patched AGB_FIRM: (Yes)
Patch ARM9 access: (No)
Set developer UNITINFO: (No)
Disable ARM11 exception handlers: (No)

**Explanation of the issue:**
When going into the DS Internet Settings, if the user holds the Select button, the Luma3DS configuration menu will show up, which is not supposed to happen in this case. The issue only affects the DS Internet Settings; launching DS cartridges, DSiWare, or the DS Download Mode don't have this issue

**Steps to reproduce:**
1. Open the System Settings and go to the Internet Settings
2. Hold Select or Start and open the Nintendo DS Connection settings
3. The Luma3DS configuration menu or chainloader will show up

tstambaugh92, 2018-08-11:
Can reproduce on 11.8 with b9s.

I'm sure this is probably known but it can also be done when exiting 3ds settings all together since it acts as a sort of reboot. In both cases it only asks for my PIN if holding the select button though.

profi200, 2018-08-11:
This is actually intended. It will force boot NATIVE_FIRM (normal boot) if you return from any DS(i) or GBA game/app.

zacchi4k, 2018-08-11:
It's not when returning, it happens when launching

profi200, 2018-08-11:
Actually read that wrong. Still, what is the issue? If you want to disable scaling hold START instead.

zacchi4k, 2018-08-11:
Well Start brings up the chainloader instead. Plus, I reported this since it shouldn't be happening, not because I needed the scaling

TuxSH, 2018-08-17:
Will be fixed later.

closed 2018-08-21 by ihaveamac