LumaTeam/Luma3DS #1121: [Feature] Top screen splashes on pin input screen

dvdcd, 2018-08-13:
Having a bottom screen splash is amazing, but i'd love a top screen one as well.
How it might work:
If your pin is 4 inputs long, then you make 5 splash images labeled something like top_pin0 etc.
Each one has a input box that fills in one character at a time, and instead of drawing a * on the screen, luma will switch from 0 to 1, 1 to 2, 2 to 3, and 3 to 4. if the pin is wrong, luma will go back and display splash 0.

This idea is kind of outlandish, but it'd be awesome to have.

tstambaugh92, 2018-08-14:
I use 8 buttons and don't think I spend more than a second on this screen. In a previous pull request or issue someone it was mentioned they didn't want a splash to possibly cover text, which would be on the top screen.

AuroraWright, 2018-08-21:
Won't be implemented because of what the above comment said

closed 2018-08-21 by AuroraWright