LumaTeam/Luma3DS #1124: System Crashes when trying to open any Emulator

Sp4rkl0n, 2018-08-17:
System model: **2DS**

**SysNAND version (+emu/redNAND version if applicable): 9.8.0-25E (Downgraded)**

**Entrypoint: b9s**

**Luma3DS version: 9.1**

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (**4**)

Splash: (Before)

PIN lock: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (**x**)

Enable game patching: (**x**)

Show NAND or user string in System Settings: (**x**)

Show GBA boot screen in patched AGB_FIRM: (**x**)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
The system crashes when trying to open any Emulator (Or trying to play the soundhax track on a compatible version)

**Steps to reproduce:**

1. Open Homebrew Launcher 

2. Start an Emulator (or try to start soundhax)

**Dump file:**

[crash_dump_00000002.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/2295601/crash_dump_00000002.dmp.zip)

Derek1199, 2018-08-17:
soundhax won't work because rosalina changes memory mappings. Also, there's no need to try running such an old exploit. You can access Homebrew just by using rosalina.

Shadow322, 2018-08-17:
Update to 11.8

TuxSH, 2018-08-17:
Rosalina/Luma3ds breaks compatibility with *hax, please follow 3ds.guide for the instructions of how to run homebrew (in a much cleaner way).

closed 2018-08-17 by TuxSH