LumaTeam/Luma3DS #1128: ARM11 , Data abort, Translation - Section, Rabbit_R error

Jaxreiv, 2018-08-18:
**System model:**
N3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.8.0-41U SysNAND
**Entrypoint (How/what you're using to boot Luma3DS):**

Bootstrap9? (I used Seedminer if that helps)

**Luma3DS version:**
Latest, 9.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x)

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Basically whenever I try to check my gear in MHXX, I get this Error 
[image]

Yes I did make a back up of the save and retried it (I used a save editor, I also use checkpoint for saves) and it still came up despite it.

I also tried it without the English patch.

**Steps to reproduce:**
If you have MHXX installed with latest update (1.4) 
1. Save edit

2.Launch save with edited items

3.Mess around in equipment box with items (Or scroll over them)

4.Crash happens.

**Dump file:**
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/2299112/dumps.zip)

piratesephiroth, 2018-08-18:
did you transfer your save data from Generations?

This will happen anytime you scroll over aa Generations-exclusive event gear. The game is japanese so it doesn't have these west-only items.

Jaxreiv, 2018-08-18:
No I did not, I only have one event gear and that event is also in MHXX and it’s right next to the rest of my weapons and I scroll over it just fine and able to equip it with no problem.

When I save edited, I edited in the Lightbreak Blade, a Raging Brachydios Great Sword which is found in the game as it was MHXX exclusive, not found in MHX/Gen.

I also edited in Deviant armor, and had it transmogified, I’m thinking it could be the transmogified armor, because I don’t have that option yet cause G-rank missions are a pain to do solo. (about 25 mins with current gear) will try that when I get home and have the chance to un-transmogrify the armor I put in to see maybe if that’s the case.

Jaxreiv, 2018-08-18:
Also, sorry for the double post but it also crashes (I removed the Sword) whenever I go over it’s slot.

closed 2018-08-31 by yuukieve