LumaTeam/Luma3DS #1130: Camera related crash

zacchi4k, 2018-08-21:
**System model:**
Tested on 3DS, 2DS, N3DS and N2DSXL

**SysNAND version (+emu/redNAND version if applicable):**
Tested on 11.6, 11.7 and 11.8 E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Tested on both Fastboot3DS and Boot9Strap

**Luma3DS version:**
Tested 9.0 and 9.1

**Explanation of the issue:**
Normally, with OFW or with older Luma versions (eg: 6.6), launching an application that uses the cameras and then trying to launch the home menu camera feature would result in a prompt that simply says the feature can't be used because the currently open application is already using the cameras; however, trying to do the same thing on Luma 9.0 and 9.1 results in a crash. This happens even if the ARM11 exception handlers are disabled, in which case the "An error has occurred, please save your data, then restart the system" prompt gets shown. 
Note that I've tested this on 2.1 too and there's no issue there 
**Steps to reproduce:**
1. Open the 3DS Camera application
2. Go to the home menu and press L+R
3. Crash

**Dump file:**
Dump files from a O3DS and a N3DS, both on 11.8E
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/2307280/dumps.zip)

profi200, 2018-08-24:
Known and reproduceable.

closed 2018-09-25 by yuukieve

TuxSH, 2020-04-19:
actually duplicate.

NightYoshi370, 2020-04-26:
What issue is it a duplicate of?