LumaTeam/Luma3DS #1136: elp An exception occurred

laser04, 2018-09-01:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Shadow322, 2018-09-01:
What's an "elp" error?

laser04, 2018-09-01:
zzz

laser04, 2018-09-01:
Sorry, I made a mistake.

yuukieve, 2018-09-01:
I'm completely guessing here, but I don't this is Luma's issue; it's possible FBI has issues with the different shared font on KOR/CHN devices; when I had a KOR emuNAND, FBI always rendered the text very weirdly.

laser04, 2018-09-01:
Oh, I tried to run FBI another version, and now it works.
Thanks!

closed 2018-09-01 by laser04

LittlestCube, 2019-01-23:
Could you please post which version you used? I have the same issue but an ARM9 error instead of an ARM11 one.