LumaTeam/Luma3DS #1137: [feature request] 300 SW limit removal

ederenzi78, 2018-09-02:
Ad time goes by, the 300 SW limit on 3DS is becoming really annoying. I don't know if it's  feasible but it's removal, at least on the New 3DS, could be a welcome addition.
Thanks!

Robz8, 2018-09-03:
and the app is now out!
https://gbatemp.net/threads/release-3dsbank-nintendo-3ds-folder-selector.516996/

TuxSH, 2018-09-05:
yeah so basically something that just swaps the name of your Nintendo 3DS folder with another one, that for some reason runs in twl mode -- if you want to do it manually, it's just that.

The problem is of course that two different folders means savedata is saved in a folder but not another, that title DB is inconsistent, HomeMenu extdata also is, etc...

As for an _actual_ fix, since this issue is a duplicate of many others, I tried to look at it long ago, and didn't find anything worthwhile to patch on Home Menu. I think the limit is possibly hardcoded in multiple places, which makes the request infeasible. 

(not to mention that I have no idea how people reach the limit with legitimate copies of games and homebrew -- and for the latter, you could just use 3dsx in most cases -- see 3ds.guide for guidance)

closed 2018-09-05 by TuxSH