LumaTeam/Luma3DS #1139: Input Redirection doesnt Reconnect and Freeze whole Devices.

Mitsukiii, 2018-09-03:
**System model:**

ALL
**SysNAND version (+emu/redNAND version if applicable):**
11.8

**Entrypoint (How/what you're using to boot Luma3DS):**

BS9

**Luma3DS version:**

9.1

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: (1 )

Splash: (- )

PIN lock: ( -)

New 3DS CPU: ( 1)

--

Autoboot EmuNAND: (1 )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: (x )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

Rosalina's Input Redirection does not Reconnect by itself and also Freeze the Device until manually Disable Input Redirection.

**Steps to reproduce:**

1. Start IR

2. Turn of WIFI

TuxSH, 2018-09-05:
Known issue. Not sure if fixable, sorry.

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH