LumaTeam/Luma3DS #1140: Sooooooo I have cfw on 11.8 and luma3ds 9.1 and when running fbi or freeshop and try to install a cia it says "an exception occurred" I tried disabling it from the luma menu but instead of the error message it just crashed help

lugia167, 2018-09-09:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

lugia167, 2018-09-09:
plz  help

lugia167, 2018-09-09:
I did I could not find anything that why I made the post

tstambaugh92, 2018-09-09:
It's not related to Luma

lugia167, 2018-09-09:
oooh ok so do you know how to fix it

lugia167, 2018-09-09:
im not saying u do no how but im 
asking if btw

tstambaugh92, 2018-09-09:
I do know how, and if this was a piracy support forum we'd all be happy to help.

lugia167, 2018-09-09:
im new here and I dont know why you cant help me because its not a piracy fourm

ihaveamac, 2018-09-09:
If you want help, you need to provide more details. And we don't assist with piracy here (freeshop).

lugia167, 2018-09-09:
ok

TuxSH, 2018-09-09:
Not a luma bug.

closed 2018-09-09 by TuxSH