LumaTeam/Luma3DS #1148: issues entering menus from Fastboot3ds

pcs3rd, 2018-09-23:
I am assuming this is for all devices, especially while chainloading luma from fastboot3ds with a payload keycombo set to select. But, it usually takes a few tries to get luma to load it's own configuration and chainloader menus.
Possible fix would be to have an option to listen for start/select, button presses, etc while splashes are being loaded/displayed.

profi200, 2018-09-25:
Yeah, this is not a bug. There is no way around this other than using different key combos.

As for the key combo detection on splash: That's something for the Luma devs to decide.

Multimegamander, 2018-09-25:
Why would you have Luma set to Select, that's like having CakesFW set to L

closed 2018-09-25 by yuukieve

Shadow322, 2018-09-25:
:)