LumaTeam/Luma3DS #1149: Add downgrading past 11.4 on n2ds?

Multimegamander, 2018-09-25:
**System model:**

[New 2DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.8.0-40U SysNAND, 11.6.1-00U DevmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[fastboot3DS]

**Luma3DS version:**

[Luma 9.1]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: (X)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Could you add downgrading below 11.4 to the new 2ds, as it causes an ARM 11 on boot? I'm not sure if it's hardware, or software preventing downgrade.

**Steps to reproduce:**

1.Downgrade to -11.4

2.Get a nice thicc brikk

**Dump file:**

TuxSH, 2018-09-25:
11.4 is the version that added support for the New2DS hardware.

closed 2018-09-25 by TuxSH

Multimegamander, 2018-09-25:
It's a generic ARM 11 crash, you can't dump that