LumaTeam/Luma3DS #1187: exceptional error has occured Arm11 (Core 1)

JayVagabond, 2018-09-26:
Hi AuroraWright, 

       I respect what you do & your reputation on the 3DS hacking scene is exceptional but help me out so I my old 3DS has the latest 11.8.0-41E & I have just installed safeb9 w/ all the boot9strap files via guide through recovery mode then of course the latest version of Luma3DS via also the guide. After I enabled hb app via rosalina menu on download play I attempt to lauch the homebrew on download play and it gives me the arm11 (core1) error, Same also w/ the steelhax I was using before I installed B9S and Luma3DS. Please help me out. Thank you. Dump files are attached for reference

Jimi-GG, 2018-10-22:
Hi,
I had the same issue as you when trying to launch the homebrew on download play. My arm11 (core1) error was due to a file incorrectly spelled. I fixed the issue by renaming the file "boot(1).3dsx" to "boot.3dsx" on the root of my SD card. I suggest you to check the name of all the files/folders on your SD card and make sure everything is correct. Hope it helps

closed 2018-10-22 by yuukieve

Margen67, 2018-10-22:
@yuukieve I believe you've mislabeled the issue. It's an _exceptional_ error :^)