LumaTeam/Luma3DS #1192: Luma unable to read sdcard

pcs3rd, 2018-10-04:
When booting via Fastboot3ds, (don' t think the issue would be effected by switching back to b9s), but when booting via sdcard, The fbi h&s inject won't boot, "swap current title with homebrew title" succeeds in rosalina, but when relaunching the injected app, a arm11 svcbreak error forces a reboot. same result with the h&s inject. If I launch an app and then reenter the home menu, it creates menu data every time. 

3ds hardware is most likely not at fault since luma boots from sd and makes a config in /luma, GM9 reads and writes just fine (I made a dummy file), my pc reads the card just fine, and fastboot3ds loads it's config from the sdcard...
im running a 16gb (micro)sdcard with an adaptor (with writeprotect disabled) on the latest firmware & luma...

james-d-elliott, 2018-10-05:
Why use the inject? If you are booting Luma3DS from Fastboot3DS just use Rosalina to launch it.

Grab this: https://github.com/fincs/new-hbmenu/releases/latest
Follow section III of this: https://3ds.hacks.guide/finalizing-setup

pcs3rd, 2018-10-05:
@noirscape github wont let me upload .bin files, so i Put the on google drive...
[Dump 1](https://drive.google.com/file/d/1sbFzsraFRza2bOXyuC6Bf5aEiR1FSR_W/view?usp=sharing)
[Dump 2](https://drive.google.com/file/d/1tOM-LJUP-C83Fv_xltceiuzKVGusYwv6/view?usp=sharing)
I am running luma 9.1 with 11.8.0-41u

james-d-elliott, 2018-10-06:
I'm not instructing you to install hblauncherloader. Follow the instructions. App injection is no longer supported.

pcs3rd, 2018-10-07:
@james-d-elliott sorry for misunderstanding. I'll try tommorow morning.

james-d-elliott, 2018-10-07:
No worries and once you follow the guide the "Download Play" icon should open homebrew.

profi200, 2018-10-08:
That doesn't sound like an issue caused by fastboot3DS at all. It's long gone when the firmware booted successfully.

I dunno what the issue is but if loader calls svcBreak() there is something seriously going wrong on your 3DS.

pcs3rd, 2018-10-09:
@profi200 i already tried reformating the card...
might be toast ( the 3ds)

pcs3rd, 2018-10-09:
@james-d-elliott slight problem, still getting svcbreak after trying to run with ds download software. could it be a formatting issue with the sdcard or could it even be degredation of hardware? I have been formatting with a chromebook instead of windows lately...

james-d-elliott, 2018-10-09:
Could be. Could also have success with another card, try another if you have it. Might be worth looking for help on the Discord over the weekend or a time you can spend a few hours with someone.

Might also be worth replacing the Luma3DS firm file, it could be corrupt. I've seen issues where files would be corrupt on SD (I believe) and seem to work, but failed in various areas and replacing the firm with the same version fixed it.

pcs3rd, 2018-10-10:
@james-d-elliot K Thanks. I will try doing another factory reset, redownloading luma, and a new sdcard. I have downgraded to a 16gb sdcard from the 32gb one i was using when this started. If it's ok with everyone, I will keep this issue thread open so I can type out any solution out on here for future users.

james-d-elliott, 2018-10-10:
I think that should be fine, not trying to avoid helping just feel you'll get better help if it's in real time.

pcs3rd, 2018-10-11:
@james-d-elliott I know. its easier instead of moving a late afternon at a time (In my Eastern/New York  timezone at least)

yuukieve, 2018-10-22:
I experienced similar strange/broken behavior when I had a damaged mSD->SD adapter/SD card itself. Unlikely to be a Luma issue.

closed 2018-10-22 by yuukieve

pcs3rd, 2018-12-11:
@yuukieve I think that Was the issue