LumaTeam/Luma3DS #1195: Input Redirection stops responding after variable soft resets

cursedtoast, 2018-10-13:
**System model:**

New 2DS

**SysNAND version (+emu/redNAND version if applicable):**

SysNAND 11.8.0

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

9.1

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I've written a shiny hunting bot for Gen 7 Pokemon games that takes advantage of the Rosalina Input Redirection.

Step forward -> If not shiny -> soft reset.

The problem is, randomly, Input Redirection stops working. And interestingly, it's always right after loading a game in Pokemon Ultra Sun/Moon. The 3ds isn't locked up itself. I can try stopping the client on the PC (currently using Release 3.0), and restarting it, with no go. Even starting a new client on a different PC I can't get it to work.

I've also tried disabling Input Redirection on the 3ds system, then re-enabling it.

The only way I can get Input Redirection to work again, is to turn off the 3ds and turn it back on and start all over again.

There isn't a specific time this happens. It can happen 900 soft resets in, it can happen 40 soft resets in. It might not ever happen at all during a hunt. Though usually, it happens around 600 soft resets in. The program I wrote literally just emulates an Xbox 360 controller. Which is still functioning fully when this issue happens.

Something is happening where Input Redirection stops responding and connections to it are no longer possible until after system restart. Unfortunately it's not consistent, but it does happen several times a day and is annoying.

**Steps to reproduce:**

1. Using Input Redirection, load your save file in Pokemon Ultra Sun/Moon.

2. Soft reset

3. Repeat steps 1-2 a variable amount of times. Eventually, upon game load, Input Redirection will stop communicating with any clients and the DS system will need to be rebooted and Input Redirection restarted in order to receive communication from any client again.

Mitsukiii, 2018-10-16:
its a common Issue with the Luma Debugger, once Wifi Dies the Game freeze, only way to recover is disable the Luma Debugger in Rosalina Menu

cursedtoast, 2018-10-17:
I don't think I ever have the Rosalina debugger enabled in the first place - does it enable when you enable Input Redirection?

Mitsukiii, 2018-10-17:
Yes

cursedtoast, 2018-10-17:
Okay, I'll take a look the next time my bot locks up. Should I just disable the debugger in the future to prevent this from happening, then? Thank you.

Mitsukiii, 2018-10-17:
If u disable it, IR won't work anymore. I would say use NTR instead

AuroraWright, 2020-11-17:
Does it happen on latest?

closed 2020-11-17 by AuroraWright

Bappsack, 2020-11-18:
yes just tested, if wifi dies the 3ds will still in a completly lock until you disable IR via Rosalina