LumaTeam/Luma3DS #1196: Luma3ds - Error

YuukiRondo, 2018-10-19:
Hey, i've got a problem using Luma3ds (It's my first time)

I've installed it, and everything goes ok, but when i try to launch the homebrew launcher my console keeps on the 3ds logo, this happens after i choose "Switch the hb. title to the current app" option.

[image]

I have a 2ds Old, thank's in advance, i hope my problem does have a solution

james-d-elliott, 2018-10-19:
https://github.com/fincs/new-hbmenu/releases/latest

Grab boot.3dsx and and place it in the root of your SD card. Replace the file if it exists. Make sure when you choose the "Switch the hb. title to the current app" option you're in an app (download play recommended). Then close the app  (home button, then x) and relaunch it. Should work fine.

yuukieve, 2018-10-19:
User error, not a Luma issue.

closed 2018-10-19 by yuukieve