LumaTeam/Luma3DS #1197: what's mean momiji？

Matthewliu7, 2018-10-23:
When I play pokemon，it always broken with momiji in luma error list.
I can't fix it,even ctrtransfer....
How can I fix it?

jonirrings, 2018-10-24:
if it is an in-game item, it the game issue, not Luma3DS.
if not an in-game item, a screenshot may be expected.

Vikijaka, 2018-10-24:
Same here Pokémon UM+o3DS, since update to Luma9.1 i got this "momiji" Error, when i am leaving/porting at my current Area (Team Magma@AetherFoundation). its possible to go through all Rooms, but i can't leave it via Entrance nor after Groudon fight via Teleporter.

[image]

noirscape, 2018-10-24:
Momiji is just the name of the process that Pokemon has.

Are any of you using no-outline patches or anything?

Matthewliu7, 2018-10-25:
> Momiji is just the name of the process that Pokemon has.
> 
> Are any of you using no-outline patches or anything?

How can I fix the problem？I was try ctrtansfer and get other ROM，but it useless..

No-outline before 1.1update is useful.

noirscape, 2018-10-25:
@Matthewliu7 - Are you using any patches for Pokemon.

If you don't know what that means, post an image of the `luma/titles` folder from your SD card here.

Matthewliu7, 2018-10-25:
Of course No patch,I just told he no-outline patch is useless...I don't use the patch...