LumaTeam/Luma3DS #1198: [ROSALINE] [FEATURE REQUEST] Autostart debugger when a specific app starts

Rajveer86, 2018-10-24:
I would like the ability to autostart the debugger when a specific app/process name starts (or if not possible, a setting to enable it by default for all "3dsx_app" processes). This would make the development process flow much better as I would be able to compile, push the binary to the 3DS and attach the debugger all in one step without having to manually open Rosalina, enable the debugger, set it up for my process, then attach e.t.c. This would save seconds, but seconds for every run is a lot of time.

Note: I was unable to post this Rosalina feature request to [#752](https://github.com/AuroraWright/Luma3DS/issues/752) as it's been restricted to colalborators, sorry for any inconvenience.

TuxSH, 2019-05-01:
Forgot to close, already implemented

closed 2019-05-01 by TuxSH

Rajveer86, 2019-05-02:
Awesome, thank you, looking forward to trying it out :) Do you know when you'll do the next release or should I build it myself?

Rajveer86, 2019-05-04:
@TuxSH I built the geoshader example and tried running it with this feature enabled in 2 different ways (pushing to Homebrew Launcher with 3dslink, and also manually starting it from the file system), however in both cases the system hangs with black screens.

Oreo639, 2019-05-04:
Did you try connecting gdb after it hangs (the 3ds will wait for you to connect gdb)?

Rajveer86, 2019-05-04:
Silly me, no I didn't and you're right, the app starts once I connect gdb to the 3DS.

One thing I noticed is that forced debugging of the next app doesn't stay on once you exit the app it's used on i.e. I would like to be able to push my binary to homebrew launcher and automatically enable debugging, then exit the app and repeat, without having to keep on enabling this in Rosalina.