LumaTeam/Luma3DS #1202: [Feature Request for Rosalina] Toggle power to Game Card slot

tamamushiiro, 2018-11-11:
Similar to toggle LED and wireless already in Rosalina, would it be possible to toggle the power for the game card slot? Mainly for those DS flashcards that drain the battery even when not in use.

(I couldn't write on #752 since I'm not a collaborator.)

yuukieve, 2018-11-11:
My fault— it was locked during a spam incident and I forgot to unlock it.

bel3atar, 2018-11-12:
do r4s drain power even while not being used?

urherenow, 2018-11-12:
Very little. I’m sure measurable though. Why don’t you test it? Fully charge, power on, unplug, and close the lid. See how long it takes to shut off empty, with a game cart, and with an R4. Compare the times.

gaiaweylyn, 2018-11-12:
I don't know about R4s, but I know SuperCards have their own CPUs for emulation purposes. If this feature were possible it'd definitely be useful for those.

urherenow, 2018-12-23:
And while I accept that result (and always thought so anyway), we can’t cut the power to the card slot. If you do that, you can NEVER sleep in the middle of a game; it will assume you removed the cartridge. You can’t remove a cartridge in sleep mode without losing your spot in a game.

McPcholkin, 2018-12-23:
I think means manually disable slot in Rosalina menu, just disable it when slot not used, to leave R4 in slot and not lost it on backpack or pocket.

Nuck-TH, 2018-12-23:
yes, this request doesn't mean developing any advances powersaving algorytms, but merely ability to disable slot1 while not in use to keep flashcart stored in it.

profi200, 2018-12-23:
There are cmds which apparently do this but testing is needed to confirm if it really cuts power to the slot.

https://www.3dbrew.org/wiki/FS:CardSlotPowerOff

coolmccoolsons, 2019-02-04:
> There are cmds which apparently do this but testing is needed to confirm if it really cuts power to the slot.
> 
> https://www.3dbrew.org/wiki/FS:CardSlotPowerOff
> 
> Update:
> Measured and confirmed it does cut off power completeley.

how do you use this?

urherenow, 2019-02-04:
Probably a better conversation for a forum or Discord. Either way, cutting power will cause problems if you cut it in sleep mode while actually running software from the card...

profi200, 2019-02-08:
> > There are cmds which apparently do this but testing is needed to confirm if it really cuts power to the slot.
> > https://www.3dbrew.org/wiki/FS:CardSlotPowerOff
> > Update:
> > Measured and confirmed it does cut off power completeley.
> 
> how do you use this?

By writing code using these 2: https://github.com/smealum/ctrulib/blob/master/libctru/include/3ds/services/fs.h#L503-L513

> [...]cutting power will cause problems if you cut it in sleep mode while actually running software from the card...

Rosalina doesn't work in sleep mode anyway. Can't reach these menu options while in sleep mode.

ParzivalWolfram, 2019-09-17:
Is there any plans to implement this? It's stupidly hard to get cards in/out of SLOT-1 on my N3DS, and i'd like to preserve what little use it has left and not have to remove my R4 outright (which takes a stupid high amount of force and scrapes the card's contacts hard) every time it's not in use to save power.

Clydefrosch22, 2020-04-02:
If it's a possible option, I would really like it as well.
My 3ds drains half its battery in 10 hours sleep mode.
I've already turned off all wifi and spotpass and whatnot too

shizwz, 2020-05-27:
Can I also add it would be beneficial to have this as an option that remains across power cycles. E.g. as non-volatile option in Luma config, with a Rosalina counterpart for on-the-go use.

For example I like to use my cart slot for games I also have installed, as a second save. E.g. I play a patched version of ultra sun and a non-patched version. Swapping between game patches off or on with fastboot 3ds key combos for NAND/SD. Oftentimes I don't even use the slot

DullPointer, 2020-07-30:
> By writing code using these 2: https://github.com/smealum/ctrulib/blob/master/libctru/include/3ds/services/fs.h#L503-L513
>
>
> Rosalina doesn't work in sleep mode anyway. Can't reach these menu options [...]

So I was getting my script-kiddy on to try and make a PoC for this, but unfortunately my game card is not recognized after switching the slot power back on. Sleep/unsleep doesn't change the recognition: only a reboot does. Any suggestions on how to force game card recognition? Also, any suggestions on how to block the power from being switched off when the card is in use? My crude solution was to block it if any title were running, but this is not ideal...

ParzivalWolfram, 2020-07-31:
That's not really a massive issue, and could come in handy for some limited
situations. I wouldn't really worry about people cutting slot power during
use as it'd be a deliberate act anyways.

DullPointer, 2020-07-31:
I guess it would be easy enough to write a warning against people cutting power in game. The downside of slot power being restored but the game card not being recognized is that you wouldn't be ab;e to set the power off by default in the Luma config settings and then toggle it back on from Rosalina when you want to play. Currently, you would either have to switch the power off via Rosalina each time you don't want to use to card and/or remember to change the Luma config on each boot to toggle it.

DullPointer, 2020-08-03:
Bad news: 
Unfortunately, I don't have the skills/time yet to implement this as a configurable feature up to the quality standards of the Luma team. I might revisit if another bit of tinkering forces me to learn how to add to the Luman config file.

Sort of good news:
I did some playing around but I couldn't find a way to force game cartridge recognition again without physically reinserting it or by rebooting. Sending the "Game cartridge inserted" notifications didn't work, probably because FSUSER_CardSlotIsInserted was already true. So there is no point having a toggle switch in the Rosalina menu. However, if the game cartridge is for DS then it will be found when booted into TWL mode (regardless of whether the slot power was switched off in 3DS mode). Therefore, a solution would be to switch the card slot power off only when a DS game cartridge is inserted and then use TWiLight Menu++ to access that DS game cartridge as needed. This wouldn't be to every user's taste though so it would need to be a configurable option...

PoC:
https://github.com/DullPointer/Luma3DS/blob/master/sysmodules/rosalina/source/main.c#L268-L276

ParzivalWolfram, 2020-08-03:
There's a TWL cart loader stub for home menus to bypass cart blacklists,
would that work? I can link the CIA if need be.

DullPointer, 2020-08-04:
> 
> There's a TWL cart loader stub for home menus to bypass cart blacklists, would that work? I can link the CIA if need be.
Is it legit to share? e.g. homebrew and not warez. If so, I could test it out. However, the PoC still might not be a satisfactory solution and it would definitely need to be a configurable option for official release.

ParzivalWolfram, 2020-08-04:
It's homebrew, yes. I'll dig it up when I can find my 3DS.

ParzivalWolfram, 2020-08-07:
> Is it legit to share? e.g. homebrew and not warez. If so, I could test it out. However, the PoC still might not be a satisfactory solution and it would definitely need to be a configurable option [...]

Backups got obliterated, but it's still online.
https://gbatemp.net/threads/twl-slot-1-launcher-first-custom-dsiware-app.414501/

DullPointer, 2020-08-07:
> Backups got obliterated, but it's still online.
> https://gbatemp.net/threads/twl-slot-1-launcher-first-custom-dsiware-app.414501/

I can confirm that that CIA works as a way to force boot into a DS cartridge, even when the card slot power is switched off in 3DS mode.

ParzivalWolfram, 2020-08-09:
Stopgap solution found, then. Is it useful for fixing Rosalina, however?

1alien1, 2021-05-26:
any news on this? i read through the whole posts here but i have no clue on how to implement the code that dullpointer used...can anybody point me on how to do it? the part about the cia i understood, so i would only have to somehow install the code in order to toggle power off. im using my dstwo in a closed case which i have to remove every time to get the card out and this would be awesome! please help :)

> 
> 
> > Backups got obliterated, but it's still online.
> > https://gbatemp.net/threads/twl-slot-1-launcher-first-custom-dsiware-app.414501/
> 
> I can confirm that that CIA works as a way to force boot into a DS cartridge, even when the card slot power is switched off in 3DS mode.

closed 2022-04-15 by TuxSH

TheNamelessWonderer, 2022-06-05:
This is great but any way to make it persist between reboots? Every time the console is restarted power is re-enabled to the cart even if you have no intention of playing it.

profi200, 2022-06-05:
Iirc if you cut power automatically via this service call on boot it will stop recognizing all gamecards completely even if you swap them. This could confuse people thinking their slot is broken.

ParzivalWolfram, 2022-06-05:
If memory serves, the DSiWare SLOT-1 forwarder works even when the slot is
disabled, so there's probably some way around that.

profi200, 2022-06-05:
@ParzivalWolfram 
I assume that doesn't even try to detect if anything is in the card slot before booting to DS(i) mode so i would not count that. I was talking about detection in 3DS mode.

JimPix1, 2023-11-18:
> This is great but any way to make it persist between reboots? Every time the console is restarted power is re-enabled to the cart even if you have no intention of playing it.

Seconded

commit c0eaf1ea (2022-04-15): Add option to toggle power to card slot.

Fixes #1202. This option is meant to be used when you have something like e.g. a flashcard that is misbehaving.