LumaTeam/Luma3DS #1206: "Data abort" ARM11 exception when opening HB Launcher

VeeTH, 2018-11-26:
**System model:**
Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.8.0-41U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
V9.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

So, I tried to install CFW by following this YT tutorial (https://www.youtube.com/watch?v=O4UDNtsYBR8) that basically follows all the same steps as 3ds.hacks.guide.

If I didn't have Homebrew already, it referred me to another video on how to homebrew your 11.8 3DS using steelhax, which thankfully worked for me. After this I swapped over to the CFW tutorial.

On the CFW tutorial, I followed all the steps and it all worked. However, after I installed Boot9Strap, this is where it went wrong. After installing it I needed to restore Download Play to its normal version, so I had to go back to the Frogminer menu through the Homebrew Launcher. When I tried launching Homebrew through Steel Diver: Sub Wars, it gave me a "data abort" ARM11 exception.

I tried disabling the exceptions in the Luma3DS config menu, but all it did was make it get stuck on a black screen when opening the HB Launcher.

**Dump file:**
[Crash dump.zip](https://github.com/AuroraWright/Luma3DS/files/2617397/Crash.dump.zip)

Oreo639, 2018-11-27:
You can't use *hax with luma. I suggest you follow the written guide for instructions on how to get to the homebrew launcher after installing cfw

VeeTH, 2018-11-27:
Ugh, I feel so dumb... I got the Homebrew launcher app now and everything else up and running after reading more on the link you posted.
This can be closed now.

closed 2018-11-27 by VeeTH