LumaTeam/Luma3DS #1213: it says this when I turn my new 2ds on An error has occurred: Unable to mount CtrNand or load the CtrNand Firm Please use an external one. Press any buttion to shutdown

IceNess1, 2019-01-07:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.9.0-43U SysNAND, 11.9.0-43U EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Via B9S
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 9.1

[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off )

PIN lock: ( off)

New 3DS CPU: (off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (Yes )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--no

**Explanation of the issue:**
It says this when I turn my new 2ds on An error has occurred: Unable to mount CtrNand or load the CtrNand Firm Please use an external one. Press any buttion to shutdown

**Steps to reproduce:**

1.

2.

**Dump file:**

IceNess1, 2019-01-07:
Any help

IceNess1, 2019-01-07:
think its because i deleted the ttles in ctr nand in gm9 is there any way i could have a title a clean one

Margen67, 2019-01-07:
If it's an issue you caused yourself it doesn't belong on the issue tracker. Stop spamming.

closed 2019-01-07 by fincs

IceNess1, 2019-01-07:
what can i do to fix this?

Margen67, 2019-01-07:
Start by reading the issue template:
# THIS IS NOT A SUPPORT FORUM! For support go here:
# Nintendo Homebrew: https://discord.gg/MjzatM8
# Rosalina feature requests go here: https://github.com/AuroraWright/Luma3DS/issues/752
# Also check the Wiki (https://github.com/AuroraWright/Luma3DS/wiki) before making an issue.