LumaTeam/Luma3DS #1216: Arm11 Excepition Error                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           Arm11-Crash-Kernel-Panic

IceNess1, 2019-01-10:
[crash_dump_00000006.dmp.zip](https://github.com/AuroraWright/Luma3DS/files/2746281/crash_dump_00000006.dmp.zip)

**System model:**

New 2DS

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.9.0-43U SysNAND, 11.9.0-43U EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

Via Boot9Strap/Sighax,

**Luma3DS version:**
    9.1

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (no)

PIN lock: (no)

New 3DS CPU: (no)

--

Autoboot EmuNAND: (no )

Use EmuNAND FIRM if booting with R: (no)

Enable loading external FIRMs and modules: (no)

Enable game patching: (no )

Show NAND or user string in System Settings: (yes )

Show GBA boot screen in patched AGB_FIRM: (no )

Patch ARM9 access: (no )

Set developer UNITINFO: (no )

Disable ARM11 exception handlers: (no)

--

**Explanation of the issue:**
i cant boot into the home menu the arm11 just pops out when i turn my new 2ds on

**Steps to reproduce:**

1.

2.

**Dump file:**

IceNess1, 2019-01-10:
any help
please

Oreo639, 2019-01-10:
Can you explain what you did right before getting the crash?

Margen67, 2019-01-10:
@IceNess1 @Oreo639 The issue tracker isn't for tech support.
Did you really have to make three issues for this?

IceNess1, 2019-01-10:
I don't know why it kept crashing on me and I installed a game in gm9 a title then the  crash happend

IceNess1, 2019-01-10:
I do know why it kept crashing on me and I installed a title on gm9  or a game for FBI it was Terraria and animal crossing new leaf welcome amiibo

IceNess1, 2019-01-15:
@Margen67 Can you help me i uninstalled b9s and now my new 2ds wont load anything well because i unstalled b9s but i still cant boot completly

Margen67, 2019-01-16:
Close this issue and go to the Nintendo Homebrew discord for support: https://discord.gg/MjzatM8

closed 2019-02-02 by 916253