LumaTeam/Luma3DS #1217: MHG generic error lumav9.1

cruzward19, 2019-01-12:
**SysNAND version (+emu/redNAND version if applicable):**

Sys 11.9.0-42U

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap

**Luma3DS version:**

9.1-bd15f51a

**Luma3DS configuration/options:**
Default EmuNAND: ( )
Screen brightness: (4 )
Splash: ( 1)
PIN lock: ( )
New 3DS CPU: (Clock + L2 )

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
Disable ARM11 exception handlers: ( )
--

**Explanation of the issue:**

MHG crashes at boot, I was using v7 of luma3ds then.
I updated luma to v9.1, I thought the problem was fixed since I was able to boot the game, however, I could only get about 3 minutes into the game.

[image]

AndyL964, 2019-02-26:
**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.9.0-42U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.3

**Luma3DS version:**
9.1-f437d923

**Luma3DS configuration/options:**
Screen brightness: 4
Splash: After payloads
Screen duration: 3 seconds
PIN lock: Off
New 3DS CPU: Clock + L2
- Show NAND or user string in System Setting
- Show GBA boot screen in patched AGB_FIRM

Thanks for reporting this. I've just about had enough with this error and was about to myself.
I've had this issue since Luma v8, I think. No matter which Luma version I update too, the error's always there. Unlike the OP, though, mine appears quite late into the game, sometimes hours or just 30 minutes into the game. From what I can tell, it seems to only happens while in the middle of a hunt.

piratesephiroth, 2019-02-26:
Are you sure this isn't a problem with the SD card? Are you using a microSD in an adapter?

closed 2019-02-26 by yuukieve

AndyL964, 2019-02-27:
> 
> 
> Are you sure this isn't a problem with the SD card? Are you using a microSD in an adapter?

I don't it is. It doesn't happen with any other game I play, only when I play Generations.
And no, I'm using a New 3DS, so just a standalone Micro SD.

piratesephiroth, 2019-02-27:
These 'generic'' errors are usually caused by problems with the SD, especially likely because it seems kind of random.

Maybe you should run an error check on it. Also reinstall the game if you're not using a cartridge (make a back up your save data and extra data first just in case)

AndyL964, 2019-03-02:
> 
> 
> These 'generic'' errors are usually caused by problems with the SD, especially likely because it seems kind of random. If it was an issue with Luma it would be deterministic, always crashing at the very same moment.
> 
> Maybe you should run an error check on the microSD. Also reinstall the game if you're not using a cartridge (make a back up your save data and extra data first just in case)

Thank you for the insight. From what you said, it must've been because my microSD was low on space, although I still had about 1 and a half GB of space left available on it.

Not too long after, I tried re-installing the game and also cleared up some space on my microSD along the way and haven't had the error since. There were no errors on my microSD as far as I can tell, so it must've been either the game was corrupted in some way or because there wasn't enough space available.

Thanks for your help! I hope this will help others who keep getting this error as well.