LumaTeam/Luma3DS #1219: ErrDisp: card removed when my SD card is cold

gingerbeardman, 2019-02-01:
**System model:**

New 3DS

11.9.0-42E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S

**Luma3DS version:**

9.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: 1(x)

Screen brightness: 3(x)

Splash: Off(x)

PIN lock: Off(x)

New 3DS CPU: Clock+L2(x)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

When my PNY brand MicroSD card gets cold the stability of Luma is compromised.

This means I cannot launch anything stored on the MicroSD card.

**Steps to reproduce:**

1. Use N3DS with Luma normally

2. Wait for temperature to drop to ~ 0'

3. ErrDisp appears

**Dump file:**
n/a

**Notes:**
This weekend I plan to return the icroSD card to the store, as it has a 2 year warranty. But before that, let me know if there are any tests you'd like me to run. I suppose that the performance of the card drops in low temperatures and falls outside of the window for Luma/timings to work correctly.

Derek1199, 2019-02-03:
This isn't a Luma issue. Furthermore, This is caused by temperature. Not luma's fault at all.

closed 2019-02-03 by yuukieve

ihaveamac, 2019-02-03:
It's Nintendo's firmware that raises this error, Luma3DS doesn't influence it in any way.

yuukieve, 2019-02-03:
Any and all non-weather-resistant electronics will falter in very frigid or hot temperatures; this is an inevitability of the way they work. Additionally, PNY is a low tier flash storage company and likely has even worse durability than more typical flash storage OEMs. However, it could also be the 3DS itself, of course; for instance, many smartphones shut off and become inoperable in frigid cold when their batteries begin to hit the 10-30% range (that percentage creeping higher and higher as the temps get colder).

Not a Luma issue.

gingerbeardman, 2019-02-03:
It's winter ⛄️