LumaTeam/Luma3DS #1220: 3DS New Black screen after deleting CFW luma with ARM9 exception

issuer423, 2019-02-14:
So my Nintendo 3DS New came up with this error after updating it to 11.9:
ARM9 data abort.
I did exactly what this video said hoping that deleting luma and other mods will restore my 3ds back to normal.
Link: https://www.youtube.com/watch?v=xzP3JlCIOA4&t=153s

Now I got black screen forever, never booting to home menu..
Any  h e l p ?

Oreo639, 2019-02-14:
If you uninstalled cfw, then this clearly is an issue with your system, not luma. 
The issues section is for posting bugs for luma. For general help, you can join the NH discord.

DMSalesman, 2019-02-15:
There's an [official procedure](https://3ds.hacks.guide/uninstall-cfw) to remove CFW. Random YouTube videos are always a bad idea, when an official method exists. Especially if that random video dates back to 2017.

Margen67, 2019-02-16:
Removing CFW is never the solution anyways, not that they would know since they clearly can't read.

closed 2019-03-13 by TuxSH

MasterGamingYT, 2019-04-05:
Don't remove CFW on a device that works, plus video guides aren't a good idea since they can be outdated, use 3ds.hacks.guide

ihaveamac, 2019-04-05:
This is an old issue.

MasterGamingYT, 2019-04-05:
Oh I forgot :(