LumaTeam/Luma3DS #1221: My 3ds isnt booting up because of luma.

watisluma12, 2019-02-19:
I believe this is an Luma error as when i was playing Mario Kart i decided to enter Mario Kart channel, when it said "Dont turn off your 3ds or get out your sd card", i got out my sd card (dont ask why) an error appeared on the touch screen, and then the 3DS refused to boot up, i couldn't even get the to the Luma settings screen, the 3ds dosent boot up, it shows an blue light, i got it to boot up days ago, but now its refusing to boot up, even without the sd card it dosent boot up is my 3DS dead?

Shadow322, 2019-02-19:
LOL

urherenow, 2019-02-19:
Correct yourself. It isn’t booting up because of YOU. Format your card and put fresh files on it, because you’ve corrupted it. And it’s YOUR fault.

jonirrings, 2019-02-19:
urherenow is right.
the sd card data if corrupted, format it or put another one in with boot.firm.

it's because of your operation, not luma.

823403553, 2019-02-19:
hhhhhhhh！ in my country we call this kind of man shadiao!

Gabriel Wang

closed 2019-02-20 by watisluma12