LumaTeam/Luma3DS #1222: 2DS SleepMode Bug

Bfbdjdi, 2019-02-25:
**System model:**
Old 2DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.9.0-xxU ("x" I don't know, maybe 39)
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
SysNAND 9.1 (The latest)
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( x)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( x)

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**

I have an 
[image]
Old 2DS. If I switch console to sleep mode, than turn DS on and open Rosalina Menu at the same moment, console will NOT exit the SleepMode. The only way is to reboot console by holding "power" button
[image]

**Steps to reproduce:**

1.
Turn ON sleep mode on Old 2DS 
2.
Turn the sleep mode OFF and open Rosalina Menu at the same time

**Dump file:**

DMSalesman, 2019-02-28:
I noticed a similar bug. On an Old3DS with Luma built as of commit `f718297591976b01d3530769d12da68636a689f6`, the console would randomly stay in sleep mode as OP described, the only way to turn it off being holding Power.

OS version: `11.9.0-42E`.

profi200, 2019-03-07:
Try to press B to exit (the invisible) Rosalina menu.

DMSalesman, 2019-03-07:
Didn't work here, because the device was locked inti sleep mode.

profi200, 2019-03-07:
Ok. Not sure if it's worth fixing a problem like that. There is no way you can accidentally hit all required buttons.

TuxSH, 2019-03-13:
Make sure you haven't activated the rosalina menu before going into sleep mode.

Additionally, this hang will happen if you have (had?) enabled InputRedirection or the gdb server. I've tried many things in attempt to fix that long-standing issue, to no avail.

DMSalesman, 2019-03-15:
@TuxSH I compiled Luma3DS v9.1-b05f9932, and was able to reproduce the bug on an Old3DS, v11.9.0-42E.

Boot it, open and close Rosalina, then close and open the lid until you get locked into sleep mode. The only way to exit it is to hold Power.

TuxSH, 2019-03-15:
Do blue light filters work for you or do they leave you with a black screen?

If it's the latter, I think I've fixed the issue ~1h ago.

DMSalesman, 2019-03-15:
Blue light filters work. I can build the latest commit and see if the bug is solved, if it helps.

urherenow, 2019-03-16:
Using latest ctrulib and Luma3ds commits, I keep getting conflicting types errors trying to compile loader.c
FSREG_LoadProgram
FSREG_GetProgramInfo
FSREG_Register
FSREG_RegisterProgram

TuxSH, 2019-03-16:
@urherenow that's normal, use latest ctrulib _release_

david4444, 2019-03-20:
Took a build from here:https://github.com/hax0kartik/luma-hourlies/releases
150-luma3ds-474eb30
Did exactly what OP said, and the 2DS was locked in sleep mode.
but when I tried pressing the B button, the system woke up.
So i guess its fixed?

TuxSH, 2019-03-22:
which commit breaks it?

Ty-Dye, 2019-03-23:
is this a o2ds only thing

TuxSH, 2019-03-24:
I can only repro it on my o2ds, I think it could be related to a bug in my custom sm reimpl (notifications and/or edge-case handling)... I need to investigate

DMSalesman, 2019-03-25:
@TuxSH The bug occurs on my New3DS as well. I could try my hand at bisecting, if it helps identifying the offending commit.

TuxSH, 2019-03-26:
Can you retry with e2ed2005095cc4d9621af5bfed5fe7271c5ac19c?

DMSalesman, 2019-03-26:
No differences, the bug exhibits the same behavior.

profi200, 2019-03-26:
Not related to this issue but i figured i should mention this: There is a bug since the sm reimplementation was introduced that makes the camera applet crash when the camera is already in use (for example by pausing 3DS camera). Normally it would just say it's already in use.

TuxSH, 2019-03-26:
urgh, edge case handling an APT. Wouldn't be surprised if it worked with official sm only because the latter's impl is so broken.

DMSalesman, 2019-03-26:
I'm available to test commits as they're out.

TuxSH, 2019-03-27:
@profi200 Fixed your bug in https://github.com/AuroraWright/Luma3DS/commit/8345f31c2a120a06d35e32bf85b5379613945968

DMSalesman: does it fix yours?

DMSalesman, 2019-03-28:
The bug is still here.

TuxSH, 2019-03-28:
What if you remove sm in sysmodules/Makefile and rebuild?

DMSalesman, 2019-03-28:
I'll try and report back in a while.

DMSalesman, 2019-03-28:
The sm sysmodule is clean. The bug is elsewhere.

TuxSH, 2019-03-28:
Can you comment this line and rebuild: https://github.com/AuroraWright/Luma3DS/blob/master/sysmodules/rosalina/source/main.c#L144 ?

DMSalesman, 2019-03-28:
You got it this time: commenting lines 144 and 150 seem to solve the bug.

TuxSH, 2019-03-28:
so the issue are the blue light filters. I think I know what's happening, hang on...

DMSalesman, 2019-03-28:
Interestingly, I tried to apply a filter and it worked fine.

TuxSH, 2019-03-28:
What if you try to comment the         screenFiltersSetTemperature(screenFiltersCurrentTemperature);
 line?

DMSalesman, 2019-03-28:
Yeah, everything works fine. Whatever it is, it's in that function.

TuxSH, 2019-03-28:
what if you increase the timeout in the above svcSleepThread call?

DMSalesman, 2019-03-29:
I replaced the 16 with a bunch of values, the most significant being 24, 32 and 64. The first and last values randomly locked the 3DS in sleep mode without even having to open Rosalina once. The middle one gave me the exact same bug we have now, if not for the slower resuming from sleep mode.

TuxSH, 2019-03-29:
What if you increase it even more, e.g to 200ms?

DMSalesman, 2019-03-29:
100ms and 200ms lock the 3DS in sleep mode as soon as I close the lid. This happens without firing up Rosalina even once.

TuxSH, 2019-03-29:
Eh. I'd have expected it to work around the issue.

Anyway, it seems like sleep mode is really finicky... the bug can be, but is not limited to, a race condition between rosalina and gsp, and it seems like stuff is timing-sensitive as well.

I think the best option here is to simply remove the "auto restore screenfilter on wakeup" feature.

DMSalesman, 2019-03-29:
It's a half victory, so let's go for it. It's been a fun ride, anyway.

TuxSH, 2019-03-29:
Thanks for your testing help :)

closed 2019-03-29 by TuxSH