LumaTeam/Luma3DS #1223: 3ds outer cameras not working

toni15538, 2019-03-01:
**System model:**

OLD 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.9.0-42U

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9strap

**Luma3DS version:**

9.1-89cc6ed

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (OFF)

PIN lock: (OFF)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( X)

Use EmuNAND FIRM if booting with R: (X)

Enable loading external FIRMs and modules: (X)

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

Disable ARM11 exception handlers: ( )

--

**Explanation of the issue:**
Only one of the outer cameras work! Can't use features such as 3d Photo, AR games etc.
The first camera works fine. It can read QR codes and it can take pictures with DSI CAMERA.
But the second camera won't turn on. When I start 3DS camera(or the camera when I press L+R in the menu), the app freezes, and when I give an AR camera to random mii in TOMODACHI LIFE, it's just showing green screen.

**Steps to reproduce:**
a. Freezing the app
1. Open 3DS camera or press L+R in the menu
2. Press inner/outer camera button
3. The app should freeze when switching to Outer camera

b. Showing Green Screen
1. Open tomodachi life
2. Give an AR camera to anyone
3. You should see green screen

**Dump file:**
None

RocketRobz, 2019-03-01:
This isn't a Luma issue.
It's possible one of your outer cameras stopped working.
There's no issues with the cameras on my console.

closed 2019-03-13 by TuxSH

gtagvr80, 2024-08-02:
yes'i know that is pretty late but because my new 3ds that the cameras don't work in 3ds app and games but bc has homebrew and has fbi and in fbi there is a feature to scan qr codes to download stuff and weirdly the caera in fbi works and im super confused.so pls reply for answers if there's any.

PabloMK7, 2024-08-02:
Hello, please ask in a homebrew related forum or in the Nintendo homebrew discord server. Those places are better suited for assistance.