LumaTeam/Luma3DS #1224: Luma 9.2?

MasterGamingYT, 2019-03-03:
I'm just asking when is Luma 9.2 is coming out

Conan179, 2019-03-03:
When it is finished :)

823403553, 2019-03-03:
Correct

Shadow322, 2019-03-03:
why do u think u need a v9.2?

DMSalesman, 2019-03-04:
@Shadow322 They might be hinting at the long list of suggestions at #752, plus the 19 commits since 9.1.

Shadow322, 2019-03-04:
@DMSalesman 13 commits and its not like they cant get off ther ass to compile a copy themselfs

DMSalesman, 2019-03-04:
@Shadow322 I know. I myself compiled a copy, but had to roll back due to a nasty sleep mode bug I'm not able to debug.

Multimegamander, 2019-03-04:
Seeing as there's a site that compiles Luma every commit, there's no reason to complain about a new release, and you can use MultiUpdater with a custom .JSON to update to each commit on the 3DS itself

RS55, 2019-03-13:
Lack of activity occurred coincidentally when deciding to do gender reassignment surgery... so nobody will know when he will return.

That or someone makes an active fork in that time and takes the list of suggestions and improvements that have been asking for a long while lol

AuroraWright, 2019-03-13:
Lack of activity occurred coincidentally also with other personal and mental health issues associated with a decreased interest in console hacking in general. The fact that there's not a net positive return for my work due to disrespectful and/or hateful users (like you) doesn't contribute positively either ¯\_(ツ)_/¯

closed 2019-03-13 by AuroraWright

TuxSH, 2019-03-13:
I'm doing some work on luma3ds, I've been doing switch stuff/being busy otherwise.