LumaTeam/Luma3DS #1225: Latest commit "a7100a1" issue

Sempiternalsharp, 2019-03-05:
I have an error after compiling the last commit of Luma3DS, when I want to use it from the CTRNand (without SD) it does not let me turn on the console and it does not access the configuration menu of Luma3DS keeping Select before turning on.

Shadow322, 2019-03-06:
fix your compiling environment because it compiles fine and works as intended

Sempiternalsharp, 2019-03-08:
> fix your compiling environment because it compiles fine and works as intended

The boot.firm from the microSD/SD memory works without problems, it stops working when you try to run from the CTRNand (without SD).

Multimegamander, 2019-03-09:
There's no need to compile Luma, get a release from the release tab, or get an hourly from https://github.com/hax0kartik/luma-hourlies/releases

Shadow322, 2019-03-09:
> 
> 
> There's no need to compile Luma, get a release from the release tab, or get an hourly from https://github.com/hax0kartik/luma-hourlies/releases

not always reliable have many times in the past these people setup a compiling bot and over time something changes in the code or compiling environment changes to where the compilers go dead because of it best to compile yourself

Multimegamander, 2019-03-09:
Kartik can fix it...

823403553, 2019-03-09:
Oh~yeah~ boy next door~

王晓辰 Gabriel Wang
+8613285319970

Shadow322, 2019-03-10:
> 
> 
> Kartik can fix it...

they always say that and it always falls threw

TuxSH, 2019-03-13:
reverted, should work now

closed 2019-03-13 by TuxSH