LumaTeam/Luma3DS #1228: error when loading homebrew launcher via downlaod play

plshelpmeimdying, 2019-03-14:
**System model:**
old 2ds
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
?? (not sure how to find)
[e.g. 11.6.0-39U SysNAND, 11.6.0-39J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
9.1
[e.g. 8.1.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off )

PIN lock: (off )

New 3DS CPU: (n/a )

--

Autoboot EmuNAND: (y )

Use EmuNAND FIRM if booting with R: (n )

Enable loading external FIRMs and modules: (y )

Enable game patching: (n )

Show NAND or user string in System Settings: (y )

Show GBA boot screen in patched AGB_FIRM: (n )

Patch ARM9 access: (n )

Set developer UNITINFO: (n )

Disable ARM11 exception handlers: (n )

--

**Explanation of the issue:**
Currently following the 3ds.hacks.guide and I am stuck on step 5, section 3- homebrew launcher. I open up download play and switch it to the hb title, but when I close and open it back up I get a screen saying that an exception occurred. Exception type: prefetch abort (svcBreak)
I double checked all previous instructions but still this happens.

**Steps to reproduce:**

1. open download play

2. Switch the hb. title to the current app

3. close download play

4. open download play
[crash_dump_00000007.zip]

**Dump file:**

[crash_dump_00000007.zip](https://github.com/AuroraWright/Luma3DS/files/2968224/crash_dump_00000007.zip)

TuxSH, 2019-03-15:
Can't reproduce. That being said there are some bugs within loader that will be fixed next release.

You should try other titles to work around the issue. H&S, system apps or games should be fine.

Nuck-TH, 2019-03-15:
@plshelpmeimdying have you waited for Download Play to fully start and load? If you do not wait - you get error, i know people who had that and waiting fixed the issue.

closed 2019-03-24 by TuxSH