LumaTeam/Luma3DS #1232: Latest luma commit (474eb30) doesn't boot

Ty-Dye, 2019-03-22:
Device info
N3DS XL
Firmware 11.9
fastboot3DS 1.2 stable
Commit 474eb30

--------------------------------------------------------------------------------------

Issue: When booting the system it just shows a black screen and doesn't do anything, though booting to Luma 9.1 dose work

-----------------------------------------------------------------------------------------

Please leave a comment if you have any questions

AuroraWright, 2019-03-22:
Apparently an issue with fastboot3DS (neither boot9strap nor godmode9 have this issue). Might be fixed, but as a reminder we only support (and test) b9s as a launcher.

derrekr, 2019-03-23:
Hi,
for future releases it would be nice if Luma could be tested on fastboot3DS as launcher as well.
We and our users would highly appreciate it.

yuukieve, 2019-03-23:
@derrekr 

If it's a Luma's bug/problem, it'll be fixed on our side of course, but we won't deliberately target or aim for fastboot3DS compatibility.

profi200, 2019-03-23:
I have pinpointed the issue to the __libc_init_array() call in ARM11 _start which in turn calls this weird _init() which got pulled in from crti.o in dkA. When i try to include a simple bx lr version it complains about:
"multiple definition of `_init'; /opt/devkitpro/devkitARM/lib/gcc/arm-none-eabi/8.3.0/armv6k/fpu/crti.o:(.init+0x0): first defined here"

> .text:1FF8008C                 EXPORT _init
> .text:1FF8008C _init                                   ; CODE XREF: __libc_init_array+3C↓p
> .text:1FF8008C                 MOV     R12, SP
> .text:1FF80090                 STMFD   SP!, {R3-R12,LR,PC}
> .text:1FF80094                 SUB     R11, R12, #4
> .text:1FF80098                 SUB     SP, R11, #40
> .text:1FF8009C                 LDMFD   SP, {R4-R11,SP,LR}
> .text:1FF800A0                 BX      LR
> .text:1FF800A0 ; End of function _init

Ty-Dye, 2019-03-23:
@profi200 could you please do a pull request or a commit for the fix

Ty-Dye, 2019-03-23:
@AuroraWright would the fix suggested by profi200's fix work with b9s to and when could his fix be applied to a new commit

commit 4e5f6e1e (2019-03-23): Fix #1232

Thanks @profi200
closed 2019-03-23 by TuxSH