LumaTeam/Luma3DS #1235: Issue with the merged pull request #1065

LPW00, 2019-03-25:
So using those hourlies from some buildbot and I noticed when I set a screen filter, my new 3ds xl on the latest firmware, eu model, fails to exit sleep mode after closing and then opening the lid. I have to force reboot through holding the power button to get my device usable again.

I'm new to GitHub and the 3ds modding scene and very inexperienced so I'm very sorry if I broke some of ur rules, because this is not a release deemed stable nor tested, or by not reporting properly, but I stumbled upon this issue and thought I'd better let u know. Please tell me if I can be of further assistance.

urherenow, 2019-03-25:
N3ds or O3ds? Did you press the B button when you thought it was stuck? Try to press it.

LPW00, 2019-03-25:
As I said, new 3ds xl. (To be fair, this is a very stupid naming scheme from nintendo) No, I didn't try pressing b because the blue led kept pulsing which is an indicator for sleep mode if i'm not wrong. But now when I was about to try it, embarrassingly I myself couldn't reproduce the issue 🤔

LPW00, 2019-03-25:
@urherenow As I said, new 3ds xl, so n3ds. (To be fair, this is a very confusing naming scheme from nintendo) No, I didn't try pressing b because the blue led kept pulsing which is an indicator for sleep mode if i'm not wrong. But now when I was about to try it (I noticed it yesterday but didn't have time to report), embarrassingly I myself couldn't reproduce the issue 🤔

Ty-Dye, 2019-03-25:
@LPW00 maybe you could try reproducing it a few more times? it may be a "sometimes" bug

Ty-Dye, 2019-03-25:
@LPW00 also what commit were you on?

also maybe your bootloader and sys version?

LPW00, 2019-03-25:
@Ty-Dye yep, after trying it once more I could reproduce it. (Pressing b does nothing, @urherenow) As I said latest firmware, eu, so 11.9.0-42E. Now bootloader... That would be boot9strap, no? Idk about commit tbh. In the luma config menu it says I'm running 9.1-4e5f6e1e, don't know if that helps. I'll check as soon as I have better internet, in about half an hour.

Ty-Dye, 2019-03-25:
@LPW00 ok thanks, when I have time ill try to reproduce this... I have fastboot3ds and a U system though

LPW00, 2019-03-25:
@Ty-Dye That'd be great. Let's hope the bug isn't region-locked🙃

Ty-Dye, 2019-03-25:
@LPW00 I attempted to replicate this 10 times (on different levels of filter to) I also tried it with power saving and wifi on and off..

could you please tell us your settings in your luma confg menu (hold select while powering on to get to the menu)

LPW00, 2019-03-25:
@Ty-Dye splash (before), clock+l2, enable game patching, and show nand or user string in system settings are enabled the rest isn't changed. I also couldn't replicate it at first but then I rebooted and tried it again some more.

DMSalesman, 2019-03-25:
@LPW00 Instead of activating a screen filter, just try opening and closing Rosalina after you boot the 3DS. Then, close and open the lid a few times to see if the console locks into sleep mode.

DMSalesman, 2019-03-25:
Possibly related to #1222.

TuxSH, 2019-03-26:
Probably due to a bug in my sm reimpl which is used for hbl, etc. (building w/o it makes my 2DS wake up from sleep succesfully).

LPW00, 2019-03-26:
@DMSalesman u're right. It appears to be unrelated to screen filters. Now I feel like I've got to apologize to @panicbit and @TuxSH... Sorry, my bad.

TuxSH, 2019-03-26:
Can you retry with e2ed2005095cc4d9621af5bfed5fe7271c5ac19c?

TuxSH, 2019-03-29:
see #1222

closed 2019-03-29 by TuxSH

MasterGamingYT, 2019-04-12:
@LPW00 If you need help in the future, considor joining the Homebrew Discord: https://discord.gg/C29hYvh. You will get more help there and that you don't have to open an issue here everytime you have a problem

LPW00, 2019-04-12:
@MasterGamingYT not to be rude, but ur comment is unasked for and also off topic. Issues get closed for a reason, take a hint.