LumaTeam/Luma3DS #1236: Trouble chainloading GodMode9 v1.8.0

d0k3, 2019-03-31:
As referenced in this issue:
https://github.com/d0k3/GodMode9/issues/466

To summarize, there are weird issues (graphical errors, black screens, shutdowns) in GM9 v1.8.0 when booted via Luma 9.1. It only happens when there is exactly one payload (GodMode9.firm) inside the Luma payloads folder and START is used to boot it.

These bugs are not reproducible any other way (chainloading via fb3ds, GM9 installed as FIRM0, multiple payloads inside the Luma payloads folder, etc...), as confirmed by multiple testers now. Using the Luma nightly instead makes the issues pop up less frequently, but it does not fix it completely.

TuxSH, 2019-04-01:
Only luma 9.1? Weird.

Is it possible for anyone to bisect. Also Luma ignores the flag because it has now way to actually deinit the screens/power down the gpu, and since the chainloader menu will init the screens...

Normally running the screeinit code twice in a row should be a no-op -- I've tested that kind of stuff with exploitation stuff

profi200, 2019-04-01:
Happens with nightlies aswell as 9.1 according to many testers.

TuxSH, 2019-04-03:
@d0k3 how's the issue on your end?

LordAensland, 2019-04-04:
Excuse my ignorance, as I am not nearly as tech savvy as you guys (just a PC tech not a programmer) so I apologize if I don't belong on this thread, but I wanted to let you know I tested the whole "only happens with the use of Start button and one payload in the folder" hypothesis and that is not correct on my end. I still am not getting complete functionality even after trying that.  I figured I'd give you some feedback as I've been trying my best to test workarounds.

I used your older version on a N3DS and 3DS about a month ago and it worked fine on both.
 Just got another N3DS today for my 11-year old godson and the Godmode9 screen logo appears and nothing else happens. This one is using New3DS with Luma 9.1, OS 11.9.0-42E, bootstrap9. FIRM0. Not sure if the model is important (I know it was with the PSP back in the day) but this one is the Golden Legend of Zelda model with the winged triforce on it.

 If I double tap the A-button and then hit the B-Button once it brings up a root directory but no other functions work (except the power off with R+Start as described in the read me file). Once I enter the root directory is where it gets extra wonky. I CAN enter the directories but I have to double-tap the A-button real fast to do so once highlighted. Now the A-button will work like normal with a single tap. If I navigate to the SD>gm9>scripts>GM9Megascript.gm9 file it once again requires a double-click-like tapping of the A-button. It will then flicker ever so briefly a code screen (or maybe an options screen... it has red text on the left and white on the right but happens faster than I can blink so I can't tell you what it says) then go to the "GODMODE9 ALL-IN-ONE MEGASCRIPT by annson24" credits page.

On the credits page I can choose options but have no way of  knowing what I choose until I try it, as I have to  count the presses of up and down on the D-pad before I press A and finally see the result. Whilst pressing up or down there is no change and it still just displays the credits. So, now the whole thing becomes one huge memory game. (Example: Pressing A brings up "Backup options:" on the screen. Pressing Down 7 times then A brings up "Scripts from Plailect's Guide". The bottom screen is completely black from the time of starting GodeMode9, so it is useless. The top screen will not highlight anything or bring up any options or such until I hit A. And no matter what I see for an option, the "GODMODE9 ALL-IN-ONE MEGASCRIPT by annson24" is always above it.). If I chose any option that has a sub-option I see just one sub-option, so if there are more I have no clue (such as Setup Luma3DS to CTRNAND).  Pressing A on a sub-option makes the system need the B-button pressed twice to back out, but short of that does not seem to do anything else.

I dropped another payload into the luma/payloads folder and changed the GodMode9.firm to X_GodMode9.firm and this issue still persists, so as I said before, it does not only happen with one payload in the folder and Start set as boot. Unless I am misunderstanding something.

Anyway, I am hoping something I said may be helpful and I am more than willing to answer any questions you have. I'll even offer my discord ID if you want to contact me directly and more in real time. And If anything I said seemed rude or such know I did not mean it to be, I joined this site to give feedback and test things and be helpful, not be a jerk. And if anything I said was against the rules I truly apologize; feel free to delete or edit as needed.

d0k3, 2019-04-04:
@LordAensland - thanks! You also need to make sure this does not happen when GM9 v1.8.0 is not chainloaded via Luma. Did you try fastboot3DS, too? You can contact me via PN (GBAtemp, GodMode9 Discord, IRC, etc) if you need help.

closed 2019-04-04 by TuxSH

d0k3, 2019-04-04:
Alright, @LordAensland - can you try the most recent Luma nightly?
https://f.secretalgorithm.com/zhoWR/boot.firm

@TuxSH just implemented a fix that should fix that once and for all.

LordAensland, 2019-04-05:
Certainly, but it will have to wait for Monday. My godson took his system to school and from there is heading to his grandmother's for the weekend.

 By the way, so, it DOES work with the older version of GM9 but you still have to update Luma after the initial install or you cannot unwrap the installed games from their present icons. However, it makes the system super delayed loading the top-screen icons (at one point none loaded at all) and powering up made the screen stay black much longer before bringing up the 3DS dashboard.

d0k3, 2019-04-14:
@LordAensland: that console is okay now, I hope? Sorry I didn't see the updated comment until now. Better to make a new comment when something new comes up, so everyone gets a notification

d0k3, 2019-04-14:
@TuxSH @AuroraWright - someone tested with the most recent Luma nightly, and the issue came up again, see here:
https://github.com/d0k3/GodMode9/issues/466#issuecomment-482354872

LordAensland, 2019-04-17:
Sorry, I was just trying not to double-post. I know a lot of forums back in the day got livid at that. Yeah, I fixed it for the most part. Some game files won't load and just crash upon start, but for the most part I got it all back into functioning order.

Sorry for the late response. Been so busy with BS I completely forgot to check this thread or even look at my PC.

commit af822e5a (2019-04-04): Fix n3ds IPS screen issues/regression, closes #1236