LumaTeam/Luma3DS #1239: Custom string not showing in system settings

MasterGamingYT, 2019-04-05:
I'm doing an actual issue this time.....
The custom string I put isn't showing in system settings, I named the txt file "customversion_sys.txt" and I put it in the root of the luma folder, but it's not showing.
Help and assistance would be greatly appreciated

MasterGamingYT, 2019-04-05:
Also the custom string I put was: LUMA3DS 11.9

MasterGamingYT, 2019-04-06:
24 hours and yet no answer

RocketRobz, 2019-04-06:
Did you enable showing custom strings in Luma3DS configuration?

yuukieve, 2019-04-06:
Believe it or not, no one's getting paid to dev this project or provide "customer service", @MasterGamingYT. 

Custom version text works for me.

closed 2019-04-06 by yuukieve

tstambaugh92, 2019-04-06:
Is Luma copied to NAND? You could try copying that file to NAND as well, I did that for my lock screen PIN I believe

MasterGamingYT, 2019-04-06:
I enabled "Show Nand or user string in system settings" @RocketRobz

MasterGamingYT, 2019-04-06:
Alright I got it working with the help of the Nintendo Homebrew Discord

tstambaugh92, 2019-04-08:
it would be helpful if you posted what was wrong and how you fixed it in case someone comes posting about it again.

MasterGamingYT, 2019-04-09:
I changed it, because I solved it